prusa3d/PrusaSlicer #2761: Wrong placement of the seam

espr14, 2019-08-14:
2.0.0 win64

No matter the type of seam selected, the location of the seam of outside external perimeter is always at the straight side, not in the curvature. But in the inner external perimeter it is placed as expected.

[image]

[image]

STL
[seam.zip](https://github.com/prusa3d/PrusaSlicer/files/3500863/seam.zip)

Jebtrix, 2019-08-14:
Wrong? You mean a feature request to specify exactly where you want a seam. Changing seam type does move it differently so not sure what you mean there.

Is this what you are trying to do? Part rotated 90, Rear seam type used:
[image]

espr14, 2019-08-14:
You're right, rear seam puts it to the rear. That works. Also random could work properly. But aligned and nearest has different policy. It should find biggest concave angle, if not available, then biggest convex angle and put it there. Straight line is not the biggest convex angle since this is in the curvature. So there's some kind of issue.

Jebtrix, 2019-08-15:
Not that its not the most optimal for printing but you can also rotate the part arbitrary degrees to have the rear seam type line up exactly where you want it.

I assume you mean like Sharpest Corner seam method in Cura behaves. With the rear seam method I never dug into this before. Never really noticed how multiple parts on the plate influenced the seam location (Nearest truly means Nearest plate wide). Choosing seam location or Sharp Corner method would be a nice addition.

Here you can see your inner perimeter influences where it appears:
[image]

Closed holes on your model to show how another objects perimeter and placement influences where it appears:
[image]

[image]

Just to be thorough throw some sharp corners in:

[image]

[image]

espr14, 2019-08-15:
Is there any documentation which would say what means each type of seam location? I though it could be the same like in Slic3r but it is not.
The rotation of the object is possible here. But it is not possible in other cases where you cannot twist two parts of the same object.

Jebtrix, 2019-08-15:
Only documentation is from Slic3r's site [Fighting-Ooze](https://manual.slic3r.org/expert-mode/fighting-ooze)

**Nearest** - This will try to choose a concave non-overhang vertex so that the seam is hidden in the inside of the concave angle. If no concave non-overhang vertices are available, it will choose a convex non-overhang vertex. If none are available, it will choose a non-overhang vertex. The choice among the candidates is operated so that the start point is the nearest to the previous extruder position. So this option will optimize for short travels.
**Aligned** - This will use the same logic as Nearest for finding the candidates, but it will choose the one that is nearest to the start point of the previous layer. This will ensure the seam is mostly aligned throughout the whole object.

Slic3r's preview window does not show you seams. You have to use a gcode viewer for that. I can tell you Slic3r behaves the same:
[image]

Jebtrix, 2019-08-15:
As for the rotating two parts of the same object that depends. If you can use right click on the multi-part and Split to Objects, then you can do right click on each object and do Add Settings > Layers and Perimeters > Seam position (scroll to bottom of list).

espr14, 2019-08-16:
OK. But we see that nearest and aligned do not pick convex vertex in this case. So there is a bug in both slicers.

bubnikv, 2019-09-16:
I checked the code, and there seems to be a bug in the placement of random seams on holes, the seams on holes are not random.

thanhtacles, 2020-08-09:
Has anyone since found out how to force the seam to be on the sharpest corner?

closed 2026-09-29 by prusaslicer-bot[bot]