prusa3d/PrusaSlicer #2766: Erroneous "Contact Z Distance"

sungdzda, 2019-08-15:
### Version
This issue is present in Prusaslicer 2.0.0 and 2.1.0 alpha. Single or MMU mode.

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
This issue only exists on the slicer end.

### Behavior
"Contact Z Distance“ does not obey user setting. While this value is set to 0.1mm by default, it is actually 0.5mm in the generated gcode. This bug significantly degrades layers above raft/support and causes the characteristic "reverse elephant foot" effect on Prusa printers. This bug renders the raft feature in Prusaslicer unusable; it also affects multi material printing, as on many occasions (for example, when using heterogeneous breakaway support) different materials' z-heights cannot be synched or are wrong/suboptimal.

lukasmatena, 2019-08-15:
I think that PrusaSlicer behaves consistently with how objects are printed. When contact z is positive, the extruder is lifted so that bottom of the extrusion does not touch the previous layer (in theory). This means there is no control of the height of the new layer and to a good approximation, the extrusions assume circular shape, whose diameter is the same as nozzle diameter. Bridge flow is applied in these cases.

So when you print with an 0.4mm nozzle, layer height=0.2mm and contact_z=0.1mm with one layer of raft, following happens:
- the raft is printed at z = 0.2mm
- the next extrusion is assumed to have height of 0.4mm
- the bottom of this extrusion should be at 0.3mm (raft layer + contact_z)
- therefore, it must be printed at z = 0.7mm

This is how PrusaSlicer works and always worked. Whether this can or cannot be done differently is another story, I'm just telling you where the numbers came from.

sungdzda, 2019-08-15:
Thanks for the helpful information. However, there is no dispute that the current Z gap between the support/raft and model is WAY too high. I don't see a way of reducing this gap via the user interface whatsoever. When I try to set the Z Contact Distance to -0.1mm, Prusaslicer gives me an error and the resultant gcode is unusable.

lukasmatena, 2019-08-15:
It should be possible to set contact z negative. If the error message mentions empty layers, it is a bug in 2.1.0-alpha1 that was already fixed (https://github.com/prusa3d/PrusaSlicer/issues/2752). 2.0.0 should allow you to do that without complains.

italocjs, 2019-12-24:
Bumping this, as i also feel that this is not ideal for all printers.  mine doenst print the first layer of the raft well

KrohmKoala, 2020-03-22:
Still no correction in last version of the slicer, it's a good slicer, not for supported print because of this issue...

kb-elmo, 2021-02-01:
same here.

setting the contact-z distance to 0.2mm on a print with 0.2mm layer height should leave out exactly one layer between the support interface and the first layer above the supports.

instead it leaves out three layers for me which results in a 0.6mm gap which is not only completely wrong in regards to the setting in the slicer but also results in mid-air printing and thus really bad surface quality.

bubnikv, 2021-02-24:
Implemented for raft with #6003 and aee136c

Object over raft is no more printed with bridging flow, but with a normal flow.
Gap between the raft and 1st object layer is configurable independently from the support gap.

Z-gap for the supports is discussed elsewhere.

closed 2021-02-24 by bubnikv