prusa3d/PrusaSlicer #2767: Please Implement High Quality Raft

sungdzda, 2019-08-15:
### Version
This issue is present in Prusaslicer 2.0.0 and 2.1.0 alpha.

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
This issue only exists on the slicer end.

### Behavior
Please implement **Cura/S3D/Zortrax** style raft. This desired raft style is characterized by thick, wide 1st layer lines and dense, smooth top interface layers. The thick 1st layer lines eliminates the impact of an uneven print surface while promoting bed adhesion. The resulted raft is of high quality and can be removed by hand easily with little effort.

Images of the desired raft implementation: 
https://i.all3dp.com/wp-content/uploads/2018/06/26144027/removing-a-raft-from-a-print-zyyx-3d-printer-180626-1284x722.jpg
https://i.redd.it/a9vdtyflsns11.jpg

Please also refer to issue #2766, which is also closely related to raft quality.

_Is this a new feature request?_
No, I would consider it a bug fix/improvement. Currently Prusaslicer places restrictions on its raft style, namely 1st layer thickness. Please make the raft more customizable.

bubnikv, 2019-08-19:
What slicer did you slice this?
https://i.redd.it/a9vdtyflsns11.jpg

sungdzda, 2019-08-21:
Hello,

Any one of Cura, S3D, or MatterControl should be able to generate raft of that style. PrusaSlicer is the only one missing on the list :)

bubnikv, 2021-02-24:
Implemented with #6003 and aee136c

Object over raft is no more printed with bridging flow, but with a normal flow.
Gap between the raft and 1st object layer is configurable independently from the support gap.

closed 2021-02-24 by bubnikv