prusa3d/PrusaSlicer #2768: Advanced settings code not adjusted for single-material prints

WarrenSchultz, 2019-08-15:
### Version
2.0.0
2.1.0a
### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Mk3S/MMU2S

### Behavior
While trying to debug tip formation, I tried doing some testing with printing only a single object in a single material, and adjusting the print parameters. I discovered that while it changes the gcode comments describing the new parameters, the actual advanced filament settings for ramming and cooling moves are unused in the actual executable gcode.

Adding a second part so that a tool change occurs resolves this issue (but increases the time and filament required for this sort of testing, as a wipe tower is generated, etc)

lukasmatena, 2019-08-15:
I'm sorry, I don't follow. Why should the advanced parameters customizing tool-changing sequence influence the gcode for single extruder print with no toolchanges?

WarrenSchultz, 2019-08-15:
It's ok, it's a bit of an edge case, but I'm not sure how the slicer caches data behind the scenes, and it's a bit hard to describe.

It *appears* that if I change a ramming/advance parameters setting while using multi-material in a profile, say, setting the cooling move speed to 1000, or the ramming curves, and then delete all the objects in the scene except for a single object (and therefore a single material), those changes I made still stick with the gcode that is generated for print, even if I drop the cooling move back down to 3. (But the commented gcode at the end of the file shows that the values have changed.)
It will continue to use the old gcode for ramming, etc, until I add another object with a second filament, then it will adopt those settings for both materials.
Does that make a bit more sense at least?

lukasmatena, 2019-08-15:
Unfortunately, no. If there is one object in the scene with one extruder, there is no wipe tower. If there is no wipe tower, there is no ramming, no cooling, etc. So even if everything is updated correctly, it will not show in the gcode, because it is simply not used. Am I missing something?

> those changes I made still stick with the gcode that is generated for print

How do you know that? What do you look at or compare to make a statement like that?

> It will continue to use the old gcode for ramming

Could you post a gcode excerpt showing ramming in a single extruder print?

I'm not saying there is not a bug. Caching and invalidating the calculation steps based on config changes is quite easy to get wrong and to forget to invalidate something. But I think that in this very case it couldn't be even noticed.

tg73, 2019-12-01:
This is a real, non-edge-case problem, for me too - with PrusaSlicer-2.1.0+win64-201909160915

I'm getting malformed tips at the end of single colour prints which cause jams. Prusa tech support told me to try changing number of cooling moves etc. Out of interest I was looking at the generated gcode for 4 vs 6 cooling moves, and it was identical except for the comments (as described by @lukasmatena), which lead me to this gihub issue.

As a point of interest, this is with Prusament PLA, not some exotic material. And I'm reliably getting jams when unloading. And I do a lot of single-colour printing. Other troubleshooting steps have been tried (eg, changing hotend PTFE).

lukasmatena, 2019-12-02:
@tg73 I don't understand. You have yourself just proved that the two gcodes are identical. If one prints and the other does not, how can that be PrusaSlicer's fault? What should PS do differently?

tg73, 2019-12-02:
@lukasmatena Both gcodes print the same, both with a tip that jams on final unload. PS should apply all the applicable settings (all toolchange parameters) to the final unload routine at the end of a print, for both MMU2S single and MMU2S multi prints. I see no reason why the final MMU2S single unload should be treated any differently from a mid-print unload during a multi - the functional requirements for success with MMU2S are the same (eg, a well-formed tip).

lukasmatena, 2019-12-02:
@tg73 Ok, I see. What confused me is that this is completely different issue than what @WarrenSchultz reported (and what I believe is not an issue at all).

You're talking about the final unload at the end of the print in MMU Single mode and that it does not respect the ramming settings etc. You are right. That unload is actually done by Printer Custom GCode, PrusaSlicer doesn't do it.

While I agree that what you're saying makes sense, I don't know how to solve it so the behaviour is consistent across all configurations that exist. I think this might be something for @rtyr to think about.

closed 2026-09-29 by prusaslicer-bot[bot]