prusa3d/PrusaSlicer #2769: [Request] Disable Part Cooling Fan During Filament Changes

nullgel, 2019-08-15:
I've noticed that when I print large things that take up a sizable portion of the heatbed _and_ have to do a filament change after only a few layers (say 4 to 8 @ 0.2mm), I have issues with the front-right corner lifting from the bed. I'm fairly certain that it is because the gantry is moved to that area for the _pause-for-filament-change_ and is cooling the part below while I take the minute (or hour, if I'm away) to do the change.

Perhaps this is done by design for some reason? Safety? It might even be okay if the Z height was just moved a lot higher when the filament changes are so close to the bed.

xoniq, 2019-08-16:
I guess PrusaSlicer is not (yet) able to triggers a `M125 - Park Head` command, but Marlin has the `ADVANCED_PAUSE_FEATURE` which has the option `ADVANCED_PAUSE_FANS_PAUSE` which turns off print-cooling fans while the machine is paused.

That sounds exactly what you need. You could turn that on in Marlin, and manually add the `M125` gcode, or PrusaSlicer might be having this as an experimental (advanced user) option to enable.

rtyr, 2021-02-03:
I believe this issue is firmware related (M600 behavior is completely handled by the firmware). Closing.

closed 2021-02-03 by rtyr