prusa3d/PrusaSlicer #2770: [Bug Report] PrusaSlicer add unexpected M900 K0 values

BastR, 2019-08-15:
### Version
_2.1.0 alpha 1_

### Operating system type + version
macOS Mojave

### 3D printer brand / version + firmware version (if known)
Doesn't matter, tested on my own designed printer and on MK3s too both with MMU2

### Behavior
This seems to be related only when using the MMU2, I don't have this problem in normal print.
Regardless of the type of filament I choose, an M900 K0 command is inserted into the gcode after the layer change, making the specific M900 K command inoperative throughout the printing process.
This happened to me on a gcode generated for a custom machine but also by resetting the application to 0 and choosing an MK3S with MMU2S.

[image]

[image]

[mmu2_10mm_filament1_0.2mm_PLA_MK3SMMU2S_5h18m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/3507029/mmu2_10mm_filament1_0.2mm_PLA_MK3SMMU2S_5h18m.gcode.zip)

Jebtrix, 2019-08-15:
Its changing to K0 before wipe tower and then changing to K30 (filament start gcode) before printing object. So K0 is not effecting your actual object.

lukasmatena, 2019-08-15:
@BastR This is a good point. The command is indeed issued before moving to the wipe tower, like @Jebtrix noted. I think that the purpose was to prevent linear advance from messing up the ramming sequence timing. The linear advance is then reset by start filament gcode.

There is a flaw though. Not all moves to the wipe tower actually feature a toolchange. In case there are empty wipe tower layers, a sparse grid is extruded with no toolchange. Linear advance thus stays disabled until the next toolchange. The M900 K0 should probably be issued before actual toolchange, not before all moves to the wipe tower.

This flaw is apparently there for almost two years (since https://github.com/prusa3d/PrusaSlicer/commit/3996535e5dea9d223fe5ad9e628bb3b841aeff63). I hopefully have a fix (https://github.com/prusa3d/PrusaSlicer/commit/0f32223ba0ec19b1e85141d0ccb997574532ad91), but it is now only in a feature branch (lm_wipe_tower_linear_advance). We'll have to test it a bit and decide if and when to merge it into master.

Thanks for the report.

BastR, 2019-08-16:
@lukasmatena  exactly, I noticed it because of a quality difference on a part of the printed part without changing the filament, it is very visible on the corners. Once the filament changes are made, everything returns to normal.

[image]

lukasmatena, 2019-08-20:
The above mentioned fix was merged into master and will be part of the next release. Linear advance is now turned off immediately before ramming and reactivated by start filament gcode that is injected after the toolchange. Sparse layers of the wipe tower with no toolchange do not turn linear advance off.

Closing the issue.

closed 2019-08-20 by lukasmatena

joaomamede, 2020-01-26:
I have this problem in 2.2.0alpha2-linux.

My wipe tower sets a M900 K0 in every tower , then I never have another M900 to put it back.
I put it in filament start gcode. that happens right before the Filament Load to be primed in the tower. I hope the M900 K0 is only needed for the unload.