prusa3d/PrusaSlicer #2771: PrusaSlicer fails to open - "Faulting module name: ig9icd64.dll" in Event Viewer.

Jsanches8448, 2019-08-16:
### Version
_2.00.0_

### Windows 10 + 1803 64-bit.
Windows 10 + 1803 64-bit.

### 3D printer brand / version + firmware version (if known)
_PRUSA / i3MK3 
Firmware = 3.7.2

### Behavior
*Upon launching the PrusaSlicer application, it opens and then immediately closes - without any errors or pup-ups.
*Unsure of the steps to reproduce but I found the following in the eventviewer:

"Faulting application name: prusa-slicer.exe, version: 2.0.0.0, time stamp: 0x5ce2dc98
Faulting module name: ig9icd64.dll, version: 23.20.16.4973, time stamp: 0x5a96b1e5
Exception code: 0xc0000005
Fault offset: 0x0000000000023944
Faulting process id: 0x4a40
Faulting application start time: 0x01d553c23ed247c9
Faulting application path: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
Faulting module path: C:\WINDOWS\System32\DriverStore\FileRepository\ki126974.inf_amd64_9168fc04b8275db9\ig9icd64.dll
Report Id: 9bc7f666-ce4e-4f9d-ab6b-e020200dced1
Faulting package full name: 
Faulting package-relative application ID:"

  *--
*Expecting the application to open and stay open.
* Actual Results = Described above.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

Is this a new feature request? = No.

Jebtrix, 2019-08-16:
Looking up that failed Intel driver brings up a lot of hits for different programs crashing because of it. Try updating that driver directly from Intel http://downloadcenter.intel.com/

rtyr, 2021-02-03:
No response. Likely resolved. Closing.

closed 2021-02-03 by rtyr