prusa3d/PrusaSlicer #2773: Some part start at second layer.

Thamone, 2019-08-17:
### Version
2.0.0

### Operating system type + version
Manjaro

### 3D printer brand / version + firmware version (if known)
Ender 5
### Behavior
After success with PLA, I tried to print this case again with PETG by using [this profile V.5](https://www.thingiverse.com/thing:3126776/comments). It is print out other things with great result but this case. When I slice the file it does not print the honeycomb pattern at the first layer but it start at second layer instead, result as it will not stick to the bed.
[image]

[image]
 
[image]

I tried to use cut function to cut buttom layer out, but it still the same.

Why it's not start at the first layer?
Is there a way to fix?

#### Project File (.3MF) where problem occurs
[RPI-BOTTOM-8020.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3511663/RPI-BOTTOM-8020.3mf.zip)

Jebtrix, 2019-08-18:
This one is kinda of a combo of settings. Though they all add to the same problem. Your honeycomb pattern in the model has 0.5mm between hole cuts. So you roughly have 0.5mm of extrusion width to work with to make that pattern. You have first layer extrusion width set to %200, so it calculates with the first layer height of 0.35 = ~0.70 (read its tooltip). Well 0.7mm is more than your 0.5mm of room thats why it doesn't show on 1st layer. Now add your elephant foot compensation of 0.1 and you just scaled that 0.5mm worth of room down more on the 1st layer.

So you have a couple options:

- Change the model so honey comb pattern is thicker spacing. ~0.7mm
- Turn on Print Settings > Quality > Detect Thin Walls. _Not my goto setting, tends to add slicing artifacts. If I was forced to use it in this case, I'd use a slab modifier just around the honey comb pattern and first layer high._
- Reduce first layer height and/or Reduce first layer extrusion width till the combined math fits 0.5mm extrude *recommended

So going with the second option, if you change First Layer Height to 0.24 you will see the honey comb sliced. Because the %200 of 0.24 = ~0.48mm fitting into the pattern width. You will also notice where it connects to main body its not quite touching. That is caused by the elephant foot reduction of 0.1. The connecting part of the pattern seems to be victim of a little decimation making it even smaller than the rest. Roughly 0.3mm from I can deduce. It may be close enough when printed though you will have decide if elephant foot compensation is worth the extra tweaking. You could also Reduce first layer height and/or Reduce first layer extrusion width further so your closer to 0.3mm extrude.

bubnikv, 2019-11-01:
PrusaSlicer 2.2 will implement adaptive elphant foot compensation, the thin parts will no more disappear due to the elephant foot compensation algorithm.

bubnikv, 2020-01-02:
The elephant foot compensation improvements are part of PrusaSlicer 2.2.0-alpha2. Closing.

closed 2020-01-02 by bubnikv