prusa3d/PrusaSlicer #2774: MMU2 extruder color selection not working

jzkmath, 2019-08-17:
### Version
2.1.0-alpha1+linux64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Ubuntu 19.04 (Linux)
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
When I try to assign colors to the 5 extruders, instead of bringing up the color selector, it selects the dropdown. It seems to be a linux-specific problem, as it never happens in the windows version.

* _Steps needed to reproduce the problem_
Run PrusaSlicer on Ubuntu or linux and try to change the filament color:
[image]
Instead of bringing up the color picker this happens:
[image]

Unfortunately this gives the user no way to change the filament color. Changing in filament settings changes it for all extruders. Unless there is a specific way to do this that I am missing here.
Also, this issue appears on version 2.0.0, and in any version (appimage or tar.bz2 form)

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2019-08-19:
Unfortunately it never worked on Linux / GTK. It works on Windows and OSX.

plampix, 2021-03-02:
A (very) dirty workaround is binding wxEVT_MOUSEWHEEL instead of wxEVT_LEFT_DOWN in the PlaterPresetComboBox constructor. This works (for me at least).

closed 2026-09-29 by prusaslicer-bot[bot]