prusa3d/PrusaSlicer #2775: !!!!! Failed to process the custom G-code template start_gcode

goneeli, 2019-08-18:
### Version
PrusaSlicer-2.1.0-alpha1+win64-201908091120.

### Operating system type + version
Windows 10 Pro 1903 18362.295

### 3D printer brand / version + firmware version (if known)
Custom Printer; Tevo Tornado, Marlin-bugfix-2.0.x

### Behavior - Unable to use a custom start G-code

- Printer Settings > Custom G-code > **Start G-code**

1. M117 Start Clean ; Indicate nozzle clean in progress on LCD
2. M107 ; Turn layer fan off
3. G90 ; Force coordinates to be absolute relative to the origin
4. G28 ; Home X/Y/Z axis
5. M420 S1 ; Enable bed leveling mesh
6. G0 X1 Y0 Z0.15 F9000 ; Move in 1mm from edge and up [z] 0.15mm
7. G92 E0 ; Set extruder to [0] zero
8. G1 Y290 E100 F500 ; Extrude 100mm filiment along Y axis 290mm long to prime and clean the nozzle
9. G92 E0 ; Reset extruder to [0] zero end of cleaning run
10. G1 E-3 F500 ; Retract filiment by 3 mm to reduce string effect
11. G1 X3 Y290 Z15 F9000 ; Move over and rise to safe Z height
12. G1 X3 Y0 Z15 F9000 ; Move back to front of bed at safe Z height to shear strings

- Plater > **Slice now**

[image]

Looking at the .gcode.tmp file, I am unable to use a custom start G-code containing G0/G1 moves because the "Variable does not exist".
[image]

Is this intentionally blocked in PrusaSlicer?  I've attempted the same start G-code in **Slic3r v1.3.0** and it sliced the stl into a complete G-code without errors as shown. Is there any workaround or downgrade version i can utilize to prevent having these errors?
[image]

Thanks!

Jebtrix, 2019-08-18:
The brackets are used for placeholder variables in Slic3r/PrusaSlicer. So the easiest thing to do is remove the [] from the comments. Use () instead ie: G0 X1 Y0 Z0.15 F9000 ; Move in 1mm from edge and up (z) 0.15mm

goneeli, 2019-08-18:
@Jebtrix That solved it! Weird that Slic3r didn't mind the brackets. Thank you

closed 2019-08-18 by goneeli