prusa3d/PrusaSlicer #2776: Feature Request: Independent fan speed for exterior perimeters

jschuh, 2019-08-18:
### Version
2.0.0

### Operating system type + version
N/A

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
I want to be able to specify an independent fan speed for external perimeters (where more cooling improves finish) versus internal perimeters, infil, etc. (where minimal cooling improves layer adhesion and part strength).

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
I looked at the CoolingBuffer and PrintConfig source, and it looks like I could code this up pretty quickly if there's any interest.

ManuGithubSteam, 2019-08-18:
Very good idea!

jschuh, 2019-08-19:
I coded up [a change here](https://github.com/jschuh/PrusaSlicer/commit/2b867b9c05f9ce8148f9702aa42486b76164cb33). I'm not sure exactly what the appropriate etiquette is (haven't contributed to anything on github before), but I'm happy to do a pull request if this is reasonable.

jschuh, 2019-08-19:
Here's a bit more context for why this change is important. CNC Kitchen recently posted [a video demonstrating how reduced cooling levels significantly increase layer adhesion](https://www.youtube.com/watch?v=Bl2ESvtBiLo) (and thus strength) of printed parts. However, doing so also negatively impacts the surface finish and resulting appearance. So, this change should allow tuning the exterior and interior to get the best of both worlds (i.e. strong layer adhesion with good surface finish).

Mithrandil, 2019-09-10:
it seems like something is wrong, according to https://www.gcodeanalyser.com/ it triggers the 100% fan speed (when External Perimeters Fan Speed is set to 100%) in the internal part and not the external perimeter
[image]

jschuh, 2019-09-15:
Thanks, I couldn't find a tool that would actually show me fan speed. I had patched the source to keep the _EXTERNAL_PERIMETER comment in and was manually reading the gcode to verify the fan kicked in where I expected. It turns out "external perimeter" isn't exactly what I understood it to be (and my gcode reading skills could use some work).

I'm going to do a little code spelunking and try to understand if the information I need is already propagated or if I will need to figure out how to do so.  Any suggestions?

jschuh, 2019-09-15:
Well, that was a fun exploration, and I think I can now authoritatively state that the fan speed rendering at www.gcodeanalyser.com is simply wrong. I reviewed the relevant Prusa Slicer code paths, and as far as I could tell my patch should work correctly since it relies on the same attributes as the logic for external perimeter speed rate. So, I then manually reviewed fresh gcode output from my patch, keying off the `G1 F####` commands to verify that the reduced feed rate for external perimeters also matched the runs where my patch adjusts the fan speed. Everything checked out, so I dug deeper.

I searched around for another slicer that would let me visualize fan speed _(should I offer to add such a visualization to Prusa Slicer?)._ I found out that www.canvas3d.io provides a fan speed visualization and allows you to set an independent perimeter speed (although it's for all perimeters, not just the external ones). So I sliced a calibration cube there with 10% base fan speed and 100% perimeter fan speed. The results show similar, wild discrepancies between the Canvas and G-Code Analyzer visualizations (images below).

Finally, I decided I should just print a calibration cube with my patch at 10% base fan and 100% for external perimeters, so I could watch and listen for the fan to spike on external perimeters. As far as I could tell it did so correctly every time—running at 10% everywhere except external perimeters, where I could hear the fan rev up to 100% (tested on my CR-10S, because the Prusa was busy printing other stuff).

So, I'm pretty sure my patch is correct and what you're seeing is just a bug in G-Code Analyzer, but perhaps there's something I'm missing?

**Visualization Discrepancies:**

[image]
_This is Canvas at layer 81 with perimeter fan speed at 100% (all perimeters, not just external) and base fan speed at 10% (notice the color coding)._

[image]
_Here's the exact same layer in G-Code Analyzer, with fan speeds that just don't make sense._

Mithrandil, 2019-09-15:
That's a great news, I will try it again then!

BTW: I'm still wondering if the higher layer adhesion on CNCKitchen is due to printing on a "still hot" lower layer, or to depositing a new layer which cools down slowly.
In the former case layer adhesion would be better for smaller parts respect to bigger parts, and bigger parts would be much les affected from having no fan at all since they have time to colld down even without the fan.

jschuh, 2019-09-15:
I've uploaded [this patch as a PR,](https://github.com/prusa3d/PrusaSlicer/pull/2922) which adds a preview for fan speeds (since I couldn't find a better way to get accurate visualizations).

Bumblebiber, 2023-02-06:
Hi, 

this seems a very cool(ing) feature to me, but I don't get how I can actually use it. Do I have to modify some files myself?

closed 2026-09-29 by prusaslicer-bot[bot]