prusa3d/PrusaSlicer #2777: Temperature settings of filament ignored / cold start

Alphaelectric, 2019-08-18:
### Version
2.0.0+ Win64

### Operating system type + version
Win10 64Bit

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 with SKR E3 DIP with TMC2208 / Marlin 2.0.x Bugfix

### Behavior
The slicer ignores filmanet temperature settings. 

On my first print with Prusaslicer I almost ran into a pretty bad situation. Just by chance, I had a bed temperature warmup in the Start GCode, but no temperature for the nozzle. 
I happened to stand in front of the machine when the printer started homing to begin printing at 80 bed temperature and 21 degree nozzle temperature!

I have since written temperature settings into the starting GCode and can print that way.

When I enter values for first and following layers in the filament section and make sure it is selected for the print, these settings continue to be ignored.
[Bug.txt](https://github.com/prusa3d/PrusaSlicer/files/3513203/Bug.txt)

* _Steps needed to reproduce the problem_
0. Set Temperatures in filament section
1. Select file
2. Slice
3. Export
4. Print

* _Expected Results_
Printer should use filament temperatures
* _Actual Results_
No temperatures in GCode except manual starting section
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
_Is this a new feature request?_
Bug

[image]

[image]

[Bug.txt](https://github.com/prusa3d/PrusaSlicer/files/3513206/Bug.txt)

lukasmatena, 2019-08-18:
The gcode you sent says

M190 S80 ; Wait for Heat Bed temperature
M109 S205 ; Wait for Extruder temperature
G28 ; Home all axes

I don't see why the printer would home at 21 degrees or what else should PrusaSlicer do.

Anyway, the temperature changing commands are inserted into the gcode only if you don't include them in the custom start gcode. Maybe this is some misunderstanding related to this. Read the tooltip on custom start gcode and go through https://github.com/prusa3d/PrusaSlicer/issues/1010 and https://github.com/prusa3d/PrusaSlicer/issues/1417 for more details.

If you are still convinced you see a bug, please enclose a 3MF with your config so we can reproduce the problem.

Alphaelectric, 2019-08-18:
Thank you for pointing me into this direction. I see where the problem is.

If I enter any temperature settings in "Custom GCode Start", they override the filament settings. When the printer started at 21 degrees, I had only a bed temperature setting noted. When it was reached, the printer took off without heating the nozzle.  

I removed all temperature settings from "Custom GCode Start" and sliced again. The correct filament settings are now entered for the first Layer and subsequently for the following layers. This will work for the unit without BL-Touch. 

With BL-Touch though, I bring up the bed and nozzle temp to 170 (no oozing) to create a mesh. Once the leveling procedure is done, I bring up the heat to printing temperature. In this scenario, I cannot make use of the filament settings. If I understand you correctly, this the point where I am using an index like in #1010, correct?

Thanks for your help in this.

lukasmatena, 2019-08-18:
If you want to heat extruder to 170 degrees, then do the mesh bed levelling and then heat to filament temperature, you should be able to do all that in the custom start gcode by doing something like
M109 S170
G28
M109 S{first_layer_bed_temperature}

The point of omitting the temperature from filament settings is to give the user absolute flexibility with the order, timing etc of the commands. If you use it in the custom start gcode, you take responsibility for setting the temperature. The placeholder variables allow you to use any config setting you want.

Alphaelectric, 2019-08-19:
I tried that and hit too errors:
1. Using the filament temperature settings: The printer heats the bed and waits for bed temperature to be reached. Once the temperature is reached, it STARTS heating the extruder, but immediately begins the print. The nozzle hits the bed cold.

2. Prusaslicer 2 does not accept the above variables in GCode. Previous variables I had in Cura don't as is the case with the example you provided. The slicer refuses to compute and throws an error.

Alphaelectric, 2019-08-19:
Hi there, thanks for the help. I managed to get the GCode start variables working!! The print should now proceed the way you explained it. 

Thanks again.

closed 2019-08-19 by Alphaelectric

Jebtrix, 2019-08-19:
You closed issue before I could finish writing but just a fyi you should move
```
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
```
to the top of your start g-code. Your doing movements before ensuring how they are interpreted. If the firmware defaults match you get away with this but its not good practice and can lead to crashing the gantry. On a CNC machine imagine it interpreting millimeters movements as inches.. not pretty.

Alphaelectric, 2019-08-19:
That's great advice! I make sure to add this to the start code. Adding code to the printer is still pretty new to me. Thank you.

Jebtrix, 2019-08-19:
One last tip, check out [http://marlinfw.org/docs/gcode/M302.html](http://marlinfw.org/docs/gcode/M302.html). You can use that feature if your firmware has it to enable cold extrusion detection.

ie
M302 P0      ; enable cold extrusion checking
M302 S170    ; only allow extrusion above 170

mzaczynski, 2020-11-18:
Great, but how to save settings to print below 170 degrees, M500 does not work, after restarting printer still is 170 minumum to extrude.

SimonSolar2C, 2020-12-14:
Some of us are still confused because there are start Gcode sections in Filament settings and Printer Settings and it seems that the Filament settings are inserted after the Printer settings.  The printer settings include the G28, G29.  Mostly the bed and nozzle temperatures are filament dependent.
For example, before I probe the bed, I want to heat the bed, and partially heat the nozzle ( so any previous nozzle drips are soft and wont hit the bed)
Here is the code I want to include in the Filament Settings:
```
; Filament gcode
M900 K0.11 ; WARNING - nothing ere to set this back after printing - restart machine if changing filament
; Set initial warmup temps
;M117 Nozzle preheat;  This all doesnt work ended up in the wrong order
;M104 S160 ; set extruder no-ooze temp;
;M190 S{max(first_layer_bed_temperature[0],80)}  ; 
; Nozzle warmup before home to avoid driving hardened ooze into PEI surface
;M109 S[first_layer_temperature] ; wait for extruder full temp before mesh bed leveling, 
;M300 S40 P10 ; chirp
```
But as I said above, it all ends up at the bottom of my initialisation.
And to add to this confusion, there are heat settings included in my Gcode - I dont know where they are coming from.(Custom machine should not have any, right?)  See lines 
[AlexNylonHeadband.zip](https://github.com/prusa3d/PrusaSlicer/files/5686541/AlexNylonHeadband.zip)

35,36 and 64,65 of attached gcode

segallis, 2024-01-12:
This is definitely a big in 2.7.1+linux-x64-GTK3.

Regardless of what is claimed about putting codes in the "Start G-code" or not, the results are inconsistent.

With an M109 code in the start code, Prusea slicer puts no nozzle temp codes, even if  the first and subsequent layers are different from the start value.  This IS A BUG.

With the M109 code commented out of the start code, Prusa inserts an M104 code for the first layer temp PRIOR to the start code - this IS A BUG.  Start code ought to be just that - the FIRST CODE.

For subsequent layers (and with no M109  in the start code), Prusa inserts an M104 to set the temp for subsequent layers.  It ought to be awn M109 code instead, else the code will not wait to reach temp, and the first subsequent layer or so may not be at the specified temp.  This IS A BUG.

With the M109 code included in the start code, Prusa will output a M140 for subsequent layers, but omit a code for the first layer.  This IS A BUG.

With a M140  heat bed code in the start code, Prusa will output an M140 code for subsequent layers, but nothing for the first.  And no wait for temp for the subsequent layers, which can take more than a minute.  This IS A BUG.

With an M190 bed temp & wait code in the start code, Prusa will suppress the initial first layer(s) bed temp, but output a bed temp for subsequent layers.  Again, this is inconsistent, so IT'S  A BUG.

In summary... 
1)  Start Codes should be FIRST - no matter what - Prusa violates this.
2)  All Prusa settings ought to generate the proper Gcode for the setting REGARDLESS OF WHAT HAPPENS IN THE START CODE.  This is the purpose of the start code - so the user can do what needs to be done before the print!
3)  Prusa handles the presence or lack of bed vs nozzle temp commands in the start code differently - this is inconsistent.  
4)  Any nozzle or bed temp command in the start code should have ZERO impact on WHATEVER the user has told the slicer to do for the first layer(s) and then subsequent layers.  Prusa should output the code for the setting the user set for the temps for first and subsequent layers regardless of the start codes!

-Greg