prusa3d/PrusaSlicer #2778: Empty layers detected, the output would not be printable

ruedli, 2019-08-18:
### Version
_2.1.0 alpha1 win64_

_201908091120_

### Windows 10

### MK3

### Behavior
When slicing atttached water.stl at 0.2mm layer height, it complains that layer z=3.4 is empty.
With layer height 0.15 there is no problem.
With the 2.0.0 released version, there is no problem

_Is this a new feature request?_
No, bug report.

#### Project File (.3MF) where problem occurs

[water_problem.zip](https://github.com/prusa3d/PrusaSlicer/files/3513236/water_problem.zip)
[image]

lukasmatena, 2019-08-19:
> With the 2.0.0 released version, there is no problem

There is. It is just not reported:
[image]
The perimeters on that layer are printed in air.

Slicing core apparently missed a layer for some reason, I have no clue why. We tried fixing your model in NetFabb but it didn't solve the issue. We'll look into it.

liftbag, 2019-08-19:
Same error type of my issue #2752. In my case it is attributable to a negative value of Z contact distance in the Support tab.

Jebtrix, 2019-08-19:
This is from a bad mesh. The model itself is missing a "layer".

[image]

Jebtrix, 2019-08-19:
Those islands are not one mesh either. If you separate to objects you can see easier:

[image]

Jebtrix, 2019-08-19:
Easiest fix is to separate to parts, then raise height of the base.

[water-fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/3516675/water-fixed.zip)

lukasmatena, 2019-10-09:
@ruedli As demonstrated by @Jebtrix (thanks), your model is composed of many parts with gaps in between. I doubt PrusaSlicer could detect and repair something like that, it cannot just merge distinct parts together (you usually don't want that).

Would you agree to close the issue?

ruedli, 2019-10-09:
Fully agree, if the error is in the data, the program should not solve it,
or it might result in undesired behaviour should you want for specific data
a correct handling. Ok was confused by the fact that the perl
implementation did not behave like this, but that might be the result of
different layer calculations and rounding.

Go ahead and close it!

(Like a lot the new features of PrusaSlicer, like range height modifiers
for the MMU2)

lukasmatena, 2019-10-09:
Ok, closing.

closed 2019-10-09 by lukasmatena

smurfix, 2019-12-04:
I disagree about closing this. The problem shows up in a *lot* of STL files on Thingiverse, and it's not always obvious how to fix them.

Couldn't the slicer simply shift everything above the gap downwards?

ruedli, 2019-12-04:
I believe if the model is wrong, any the fix should be towards making the
model correct in the first place. Shifting and combining layers above the
gap might be the right thing to do, but this should be decided by the user
in a "repair" of the STL action.

This would then be a "new feature". I see that best done in a program
specialized in manipulating stl's. Not so much the slicer.

Opinions?

dorfman2, 2019-12-22:
I agree. Yes there are issues with models, but very often it will slice fine in Cura, but not slice correctly in PrusaSlicer. Why would this be the case. Here's screenshots and the STL.
Cura v3.6.0
PrusaSlicer v2.1.1

[87619.zip](https://github.com/prusa3d/PrusaSlicer/files/3992290/87619.zip)
[image]
[image]

lukasmatena, 2019-12-22:
@dorfman2 There are also cases where PrusaSlicer is ok and Cura is not. If the user fails to provide a valid model, slicer cannot guarantee to process it "correctly". Both applications use different algorithms to make sense of such faulty models, that's why the results are not the same.

We will likely improve the model repair function in PrusaSlicer 2.3. Until then, Netfabb should repair it for you.

dorfman2, 2019-12-22:
@lukasmatena Thanks for response and explanation. I was pulling my hair out a bit (since I get better prints with PrusaSlicer).

pjratl, 2019-12-30:
this error also occurs with this model https://cults3d.com/en/3d-model/game/dragon-chess-dragon-horse-the-knight in Prusa slicer but not in Cura

dorfman2, 2020-01-02:
@pjratl While in the plater view, try right clicking the object and clicking "Fix through Netfabb." That is what @lukasmatena was referring to above that solved my issue.

bubnikv, 2020-01-02:
> Yes there are issues with models, but very often it will slice fine in Cura, but not slice correctly in PrusaSlicer.

Sometimes it is vice versa.

We are collecting these models, we are marking these issues with a "model import" tag. One day we will return to this issue, try to find a common pattern on those files that are currently not processed correctly and we will try to improve the slicing algorithm to handle these cases. The problem is, that fixing the algorithm for one class of model errors may break the algorithm for another class of errors. That's why it takes a considerable time to improve and retest the algorithm.

pjratl, 2020-01-02:
I did try model repair it did not fix the problem

this isssue in this model is the very top I sharp point ....

ForsakenNGS, 2020-01-12:
> I did try model repair it did not fix the problem
> 
> this isssue in this model is the very top I sharp point ....

I've had the same issue, too. I worked around it by adjusting the layer height for the most top layer (made it thicker). Still for the most top layer there should be an exception.

cfelicio, 2020-03-04:
Under Print Settings -> Layers and Perimeters, you can try to disable "Detect Thin Walls". In some instances this can make the object printable, and give an output similar to Cura.

tedder, 2020-04-01:
My "slicing fail" example comes from [the popular Notre Dame model](https://www.myminifactory.com/object/3d-print-notre-dame-de-paris-cathedral-91899). The top of the spires get skinny enough that they disappear before printing the 'ball' at the top. This model is 100x120mm, so it's not like I'm trying to print it super-tiny or anything.

[image]
[image]

dorfman2, 2020-04-01:
@tedder My Workaround for now has been to use a free trial at https://makeprintable.com/ . You can get a 30 day free trial (credit card required, but you can cancel without paying).

fiveangle, 2020-05-03:
Here is another example, generated from the included OpenSCAD source, if they are of any interest or use...
[Tesla Model 3 screw-on wheel centercaps.zip](https://github.com/prusa3d/PrusaSlicer/files/4570963/Tesla.Model.3.screw-on.wheel.centercaps.zip)

RFgermany, 2020-06-28:
I had a similar problem. 
PROBLEM:
In Cura it had no problems, but the print didn't look as good as on Prusaslicer. On the other hand prusaslicer did miss partially some layers.
SOLUTION:
I exported the Part as 3MF istead of STL and that fixed it!  
Maybe it helps to reduce the polygon size too.

lukasmatena, 2020-07-23:
The empty layer error will be turned into a warning in PrusaSlicer 2.3.0, so that the users can continue exporting the gcode when they see it is a minor problem (such as the Notre Dame from @tedder and similar).

ozzyman, 2020-10-30:
All we really need is the ability to either:
- turn the error checking off for this error, or
- ignore the error.

The problem I have is that I'm trying to print a model with deliberate empty layers:
[0.08_vertical_clearance_test.zip](https://github.com/prusa3d/PrusaSlicer/files/5468027/0.08_vertical_clearance_test.zip)

The model IS printable, because I have already printed it using Snapmaker Luban. What I want to do is use the variable layer height feature in Prusaslicer to speed up the print instead of printing the whole thing at 0.08mm (Luban is very basic and does not have variable layer height). Obviously this is a test model. The model I want to print using vertical clearances is much bigger and will take days to print at 0.08mm layer height. 

Prusaslicer will not allow the gcode to be saved because there is an error. I want to tell it to ignore the error and export the gcode anyway.

lukasmatena, 2020-10-31:
See https://github.com/prusa3d/PrusaSlicer/issues/2778#issuecomment-662990456