prusa3d/PrusaSlicer #2780: [FR] Configurable minimum area threshhold for gap fill.

FHeilmann, 2019-08-18:
### Version
_Version of PrusaSlicer used goes here_
2.1.0-alpha1

version_2.1.0-alpha1-20-g9bd65360c

### Operating system type + version
_What OS are you using, and state any version #s_
Any OS

### 3D printer brand / version + firmware version (if known)

Any Printer
### Behavior
* _Describe the problem_
I am trying to print a part (a plastic 80T pulley) that has a GT2 tooth profile on the outside. With gap fill enabled, this part leads to lots of small deposits in the tooth tips and travel moves between them, resulting in visible artifacts and 'degradation' of the tooth profile, as these short travels usually don't trigger retracts, which can lead to small strings inside the pulley teeth.

Moreover, these minuscule deposits don't really contribute to the usability of the gear and only add to the print time and the aforementioned artifacts.

[image]
[image]
I experimented a bit with gap fill and the detect thin walls setting, as well as different extrusion widths for external and internal perimeters. While I was able to get rid of most of the travels and small deposits, this lead to some other issues with the model.

I then dove into the PrusaSlicer code and looked for places where I could prevent these small deposits from being generated and ended up in [PerimeterGenerator.cpp#L112](https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/PerimeterGenerator.cpp#L112). I found that when I adjusted these lines to:
```c
append(gaps, diff_ex(
  offset(last,    -0.6 * distance),
  offset(offsets,  0.6 * distance + 10)));  // safety offset
```
The small deposits in the tooth tips went away:
[image]
[image]

Note that the profiles and settings were identical for all resulting slices shown in this feature request. 

Unfortunately I am not familiar enough with the PrusaSlicer code base to relate the values passed to `offset` to any real metric, in order to make this configurable and submit a PR, hence why I am submitting this as a feature request:

My suggestion is a `gap_fill_minimum_area` config parameter. Gaps below this threshold will not be generated, while gaps with an area above the threshhold get filled remain untouched and get generated like the current master code does. This would allow me to fix 'problems' like the one I currently have by simply setting an area threshhold above what constitutes the area inside the tooth tips.

_Is this a new feature request?_
Yes

Thank you for your consideration,

Florian

Jebtrix, 2019-08-18:
Applicable to issue request [#2691](https://github.com/prusa3d/PrusaSlicer/issues/2691)

paulnewall, 2020-05-11:
I have the same problem, ie that tiny gap fills cause blobs that disturb the following layers. I also looked at the code. It took a long time to get it to compile (solved by using ninja, instead of cmake as suggested in the doc file). I think the place to modify is in ExPolygon.cpp function ExPolygon::medial_axis, around line 259. 
&& polyline.length() < max_w*2) { 
ignores gap fills with length less than 2 x their width. That means that narrow fills can be very short. I think it would work better with a length limit based on the general extrusion width rather than the width of the actual gap fill. I am experimenting with this.
&& polyline.length() < (max_width+min_width)) { 
which I think would be about 2x the width of a typical gap fill, and that seems to work on my current project (prusa rc3 face shield). Even better would be to add a user adjustable factor.

dartrax, 2021-09-27:
I also have this problem. Coming from Simplify3D, I'm used to its "Minimum extrusion length" [mm] setting for single extrusions. It works quite well.

closed 2026-09-29 by prusaslicer-bot[bot]