prusa3d/PrusaSlicer #2782: Dont support bridges

Shama71, 2019-08-19:
### Version
PrusaSlicer 2.1.0-alpha1
### Operating system type + version
Windows 10 x64
### 3D printer brand / version + firmware version (if known)
### Behavior
"Don't support bridges" option is on.
After slicing this result.
[image]
Аirst bridge right bottom following counterclockwise.
1 bridge length = 11.5
2 bridge length = 11.6
3 bridge length = 13.9
4 bridge length = 14
#### Project File (.3MF) where problem occurs

[Dont support bridges.zip](https://github.com/prusa3d/PrusaSlicer/files/3515587/Dont.support.bridges.zip)

Shama71, 2020-01-18:
Option "Don't support bridge" not work for 2.1.1 and 2.2.0 alpha1 and alpha2 versions.

bubnikv, 2021-02-24:
The "bridges" are not being anchored, thus they are not technically bridges. The bridges for the right bottom instance is not left to right oriented because there were no anchors found for them.

[image]

If you decrease number of perimeters from 2 to 1, then the bridges are correctly detected and supports are not generated.
[image]

closed 2021-02-24 by bubnikv