prusa3d/PrusaSlicer #2783: [FR] Make speed print speed estimation after slicing (+model recoloring) optional or abortable

nouser2013, 2019-08-19:
### Version
PrusaSlicer-2.1.0-alpha1+win64-201908091120

### Operating system type + version
Windows 10 (build 18362)

### 3D printer brand / version + firmware version (if known)
Trilabs DeltIQ XL

### Behavior
After slicing a model, which usually completes within seconds, I get to see a print preview with perimeters in yellow, infill in red, and support and brim in green. So far so good.

But then a follow-up process starts where (I guess) the slicer GCode is taken apart into various trace types, such as bridge, outer perimeter, inner perimeter, support, etc.

Usually, this second party takes 2 orders of magnitude longer. Quite cumbersome when you just want to have a quick look at the generated supports and have to wait up to 5 minutes until the slicing finishes completely.

Also, this second part cannot be interrupted, the GUI just freezes with No Response from App in windows title bar.

lukasmatena, 2019-08-19:
> the slicer GCode is taken apart into various trace types

The gcode is actually being generated at that time, that is what takes long. It is not about visualization. The gcode generation usually finishes in reasonable time. At least unless you enable 'avoid crossing perimeters' or similar features, that can increase the time a lot.

It should be possible to stop the process at any time and GUI should stay responsive at all times. Providing a 3MF like the issue template suggested would allow us to investigate more.

nouser2013, 2019-08-19:
> At least unless you enable 'avoid crossing perimeters' or similar features, that can increase the time a lot.

Ok, I get it, and yes, that was it. Can I then rephrase the FR into "add button to preview support and perimeters and infill without slicing"? I'm not sure how you would describe such a feature, as in my view, there is already layers and infill and support stuff visible:
[image]

PrusaSlicer seems to start slicing right away, if:
* I select the preview while being in the 3D edit mode
* I press the "Re-Slice now" menu command or use the hotkey CTRL+R

And as I said, the shown preview happens in seconds, the slicing takes 30 seconds (w/ crossing avoidance option disabled) or ~5 minutes with (w/ crossing avoidance option enabled).

> It should be possible to stop the process at any time and GUI should stay responsive at all times. 

Unfortunately, no. As I said, if I during slicing of a large model e.g. abort the slicing or do an interaction in the 3D edit view, the GUI becomes unresponsive immediately, until slicing was aborted or has finished successfully. If I try to resize the window or move it, windows would display "(No Response)" in the title bar of the program after a few seconds.
[image]

lukasmatena, 2019-08-20:
What you are describing is not correct behaviour. Is there anything special about your setup. E.g. don't you have your user data directory redirected into some remote and slow to access location or something similar? Does this happen with any model or just this one? Could you please share a 3MF so we can try to reproduce the problem?

You can also try running PrusaSlicer from a console with environment variable SLIC3R_LOGLEVEL set to 9 (`set SLIC3R_LOGLEVEL=9`). This will enable verbose logging that could give us a clue about what's happening.

Thanks.

nouser2013, 2019-08-20:
> Is there anything special about your setup. E.g. don't you have your user data directory redirected into some remote and slow to access location or something similar?

Not that I can think of. The local M2-SSD is connected to the PCIe with 4 lanes and does 3.4GiB/s random writes :)

> Does this happen with any model or just this one?

Yes. Scaled the attached EuroFighter to the delta's 25cm diameter and some 30cm height. Same behaviour. After pressing "abort" shortly after starting the slice, window freezes for 20 something seconds.
[EurofighterTyphoon.zip](https://github.com/prusa3d/PrusaSlicer/files/3522653/EurofighterTyphoon.zip)

> You can also try running PrusaSlicer from a console with environment variable SLIC3R_LOGLEVEL set to 9 (set SLIC3R_LOGLEVEL=9). This will enable verbose logging that could give us a clue about what's happening.

Did that, but where do I find the logfile on Windows 10? Thanks.

lukasmatena, 2019-08-20:
I managed to reproduce the issue with unresponsive UI when avoid crossing perimeters was enabled. We should fix that. Does it also occur with that option disabled? If so, could you enclose a 3MF that shows the problem, not just the STL? The culprit will likely be something else in your config (and 3MF contains full config, not just the model).

> Did that, but where do I find the logfile on Windows 10

It is printed to stdout/stderr, i.e. into the console you ran PrusaSlicer from. But since I already reproduced it, it is probably not necessary to do.

nouser2013, 2019-08-23:
> Does it also occur with that option disabled?

Will test tonight.

> If so, could you enclose a 3MF that shows the problem, not just the STL? The culprit will likely be something else in your config (and 3MF contains full config, not just the model).

Ok, in case it occurs there as well, I'll upload the 3mf.

nouser2013, 2019-08-23:
> Does it also occur with that option disabled?

Tried it a moment ago: no, I can abort _right away_ with the "crossing..." option _disabled_. Then I guess this can be considered a bug report, not an FR.

bubnikv, 2019-08-25:
Aborting issue of the "avoid crossing perimeters" feature has been fixed with 3addae3bb41862cbe27a5f04f626d0b711729662, and the feature is now also significantly faster. Closing.

closed 2019-08-25 by bubnikv