prusa3d/PrusaSlicer #2784: Excessive external_perimeter_extrusion_width error

francisco-pereira, 2019-08-19:
Version 2.1.0-alpha1+linux64
Running on Linux Ubuntu 19.04
Settings adjusted for the Original Prusa i3 MK2.5

When setting the extrusion width with a large value, the following error appear:
Excessive perimeter_extrusion_width=0.85 mm to be printable with a nozzle diameter 0.4 mm

I know from experience that 0.85mm and even a little more is printable with a 0.4mm nozzle, so this error is really limiting the printer usage.

There was no such error on previous versions of Prusa Slicer.

Jebtrix, 2019-08-20:
They added a 2x nozzle diameter limit. This should be a warning instead for sure. You can't predict every use case and clever workarounds with settings. Soft capping that for all printers that can be used with PrusaSlicer isn't a good idea at all.

bubnikv, 2019-08-20:
@Jebtrix 
> They added a 2x nozzle diameter limit. This should be a warning instead for sure. You can't predict every use case and clever workarounds with settings

I don't quite agree with you. Please understand, that "clever workarounds" will spread as a fire on the facebook, then somebody will break something, then there will be a flood of posts on github, and somebody will have to do the customer support for that "clever workaround" issues.

I asked at #2715 for a reasonable top threshold. I am asking again.

> Excessive perimeter_extrusion_width=0.85 mm to be printable with a nozzle diameter 0.4 mm

By the way, this seems to be the same number as in #2715. Where did you get the idea / number from? How is the number justified?

Zemistr, 2019-08-20:
https://youtu.be/OnxFUF3Exis

Zemistr, 2019-08-20:
Just and example about the result of printing 0.85 wall with 0.4 nozzle.
For a vase mode perfect. :)

https://youtu.be/OnxFUF3Exis

Zemistr, 2019-08-20:
@bubnikv Honestly... I just changed 0.4 to 0.8 😉

francisco-pereira, 2019-08-20:
@bubnikv 

> "clever workarounds" will spread as a fire on the facebook, then somebody will break something, then there will be a flood of posts on github, 

I think that if a "clever workaround" becomes popular, it should be seen as an opportunity for Prusa Research to turn it into a feature on it's printers.

>and somebody will have to do the customer support for that "clever workaround" issues.

A warning should be enough to inform the user that he can break something on his printer.

>By the way, this seems to be the same number as in #2715. Where did you get the idea / number from? How is the number justified?

Nothing special about this number, I've heard of people using 1.2mm with good results.
It just happens that this over extrusion is great for vase mode printing.

Jebtrix, 2019-08-20:
@bubnikv Looking at [#2715](https://github.com/prusa3d/PrusaSlicer/issues/2715) I can see hundredths and thousandths of a percentage value causing float-point precision problems. I let it sink in. I understand the motivation with these new limits to cover extreme min/max. It makes sense on multiple fronts. Your trying to mitigate _mistakes_. I'm just concerned _unusual_ values to be considered _**extreme**_. Hard coded limits should be very liberal.

I looked into @francisco-pereira's 1.2mm comment and indeed people have had success with it. So be liberal, known unusual @ 300%, so _extreme_ should considered a bit higher, no?

bubnikv, 2019-08-21:
300% is a fair number. Implemented with 7c0c5705df1b6d7739c0d20b713e99741d91cc88.
Closing.

closed 2019-08-21 by bubnikv