prusa3d/PrusaSlicer #2786: Exporting G-code immediately after slicing leaves tags in filename

roesel, 2019-08-20:
### Version
2.0.0+

### Operating system type + version
MacOS Mojave 10.14.6

### 3D printer brand / version + firmware version (if known)
Prusa / i3 MK3S + 3.7.1

### Behavior
* If you press "Export G-code" fast enough after slicing, the suggested name includes a `{print_time}` tag instead of actual print time.
* Steps needed to reproduce the problem:
  * Add an .STL, press "Slice now"
  * Immediately (still during slicing as far as I can tell) press "Export G-code"
* Current Results
  * The suggested name should now say `myprint_0.2mm_PLA_MK3S_{print_time}` instead of `myprint_0.2mm_PLA_MK3S_3h27m`
* Expected Results (any of the following)
  * The button "Export G-code" should not be clickable before slicing is finished completely 
  * If "Export G-code" has to be clickable early, the dialog should only appear after it also knows the print time.
  * If the dialog also has to appear immediately, the name of the .gcode file should simply not include the time estimate. 

This is not a feature request, however, it is complete nitpicking of a hopeless perfectionist.

#### Project File (.3MF) where problem occurs
[plater.amf.zip](https://github.com/prusa3d/PrusaSlicer/files/3520689/plater.amf.zip)

Thank you for your amazing work!

bubnikv, 2020-02-04:
The suggested name with the placeholders is finalized when saving the file on disk. Did you experience an issue when the placeholder was actually stored as part of a file name?

lukasmatena, 2023-11-21:
I have tested that the filename is finalized correctly when the file is saved on disk (tested now in 2.7.0-rc1, but it was most likely always like that). Closing.

closed 2023-11-21 by lukasmatena