prusa3d/PrusaSlicer #2789: Coredump with no DISPLAY

jasontibbitts, 2019-08-20:
### Version
2.0.0

### Operating system type + version
Fedora 30

### Behavior
```
ld00:~❯ DISPLAY= prusa-slicer
[1]    8887 segmentation fault (core dumped)  DISPLAY= prusa-slicer
```

Obviously the program isn't particularly useful when run with no options and no ability to display graphics, but perhaps it shouldn't segfault and trigger crash reporting and such.  Here's the mini-backtrace; I recognize that this may be an issue with Wx that you can't fix, in which case I'll try to at least detect an empty DISPLAY variable in the shell wrapper we have to use.

```
                Stack trace of thread 8887:
                #0  0x00007f52f882c384 _ZNK16wxAppConsoleBase17GetAppDisplayNameEv (libwx_baseu-3.0.so.0)
                #1  0x00007f52f85e7295 _ZNK8wxLogGui8GetTitleEv (libwx_gtk3u_core-3.0.so.0)
                #2  0x00007f52f85e77b8 _ZN8wxLogGui5FlushEv (libwx_gtk3u_core-3.0.so.0)
                #3  0x00007f52f88cdec6 _ZN5wxLog15SetActiveTargetEPS_ (libwx_baseu-3.0.so.0)
                #4  0x00007f52f88bfafd _Z7wxEntryRiPPw (libwx_baseu-3.0.so.0)
                #5  0x0000563d8c9c5a15 _ZN6Slic3r3CLI3runEiPPc (prusa-slicer.wrapped)
                #6  0x0000563d8c9b5f85 main (prusa-slicer.wrapped)
                #7  0x00007f52f7493f33 __libc_start_main (libc.so.6)
                #8  0x0000563d8c9c0f8e _start (prusa-slicer.wrapped)
```

bubnikv, 2019-08-21:
I was already looking a bit into this. The X server connection is IMHO abstracted away by wxWidgets, there is even no way to set the X display programmatically apart from setting the environment variable.

> in which case I'll try to at least detect an empty DISPLAY variable in the shell wrapper we have to use.

That may be the only option without modifying the wxWidgets.

rtyr, 2021-02-03:
Can we close this issue?

lukasmatena, 2022-04-05:
The empty DISPLAY env variable is now checked and the application should terminate gracefully (https://github.com/prusa3d/PrusaSlicer/commit/a09b4ff5c8bca26de24ad8afd4d531018733b0be). Closing.

closed 2022-04-05 by lukasmatena