prusa3d/PrusaSlicer #2790: False error: "extrusion_width=0.15mm is to low to be printed at a layer height 0.15mm"

Schrunch, 2019-08-20:
### Version
PrusaSlicer 2.1.0-alpha1
Windows 10

Why this release shows this error : "extrusion_width=0.15mm is to low to be printed at a layer height 0.15mm" ? I have set the nozzle dia. to 0.15mm because I am using a 0.15mm nozzle of course. I attach the screen shot of the error, the .stl file and the .ini as zip files
I am printing hundreds of fine scale miniature things with a 0.15mm nozlle with this setting for over 6 month with Slic3r 1.41.3+win64 and have excellent results. Please can use just put a warning if you don't want to remove the message, so I can use this new version.

[image]
[STL.zip](https://github.com/prusa3d/PrusaSlicer/files/3522585/STL.zip)
[Cloture 1_87 MK3 Buse0.15 PLAPlus 220_220.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/3522592/Cloture.1_87.MK3.Buse0.15.PLAPlus.220_220.ini.zip)

Jebtrix, 2019-08-20:
extrusion_width_min <= layer_height then no joy

You can't do that anymore. Use 0.151 layer heights

Schrunch, 2019-08-20:
Why I can`t, it works fine with 1.143.3 and older?

 
<https://github.com/prusa3d/PrusaSlicer/issues/2790?email_source=notificatio
ns&email_token=AHVXKKJJP6H6HAZBCUIA7C3QFRWAFA5CNFSM4IN5CBVKYY3PNVWWK3TUL52HS
4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD4X3IVA#issuecomment-523220052> ,
RWAFANCNFSM4IN5CBVA> .
<https://github.com/notifications/beacon/AHVXKKOHT7CWG6N33E5LWD3QFRWAFA5CNFS
M4IN5CBVKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD4X3IVA
.gif>

Jebtrix, 2019-08-20:
I'm not a fan of by the book limit values chosen either. Its still alpha so we just have to petition for some limit relaxing. Some fudge factor room.

Schrunch, 2019-08-20:
Thanks for your answer. I am new to github. How can we petition for some
limit relaxing the limits?

 
<https://github.com/prusa3d/PrusaSlicer/issues/2790?email_source=notificatio
ns&email_token=AHVXKKNWEBVE33KB66ZND4TQFR3NPA5CNFSM4IN5CBVKYY3PNVWWK3TUL52HS
4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD4X6BPA#issuecomment-523231420> ,
R3NPANCNFSM4IN5CBVA> .
<https://github.com/notifications/beacon/AHVXKKPB2PGBZ2APQGUQAOLQFR3NPA5CNFS
M4IN5CBVKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD4X6BPA
.gif>

Jebtrix, 2019-08-21:
Your doing it right now :)

Jebtrix, 2019-08-21:
You can also comment on other related issues or just give a post thumbs up.

bubnikv, 2019-08-21:
Extrusion width equal to extrusion height equals to a round extrusion. How well does a round extrusion stick on top of a round extrusion?

Schrunch, 2019-08-21:
A nozzle has a flat surface around his hole. The slicer should deliver enough filament to have a flat surface on the extrusion when flattening by the flat surface of the nozzle.
The sides will have a lot bigger diameter because of the compression.

screwyluie, 2019-08-21:
it doesn't matter what anyone thinks about if it'll work or not. If the program is in expert we should be able to do whatever we want with the settings. 

This error is just a warning it removes the ability to slice at all, and there's no way to turn it off.

whoever made this change needs a kick in the seat.

fix this.

lukasmatena, 2019-08-22:
@screwyluie There are two reasons additional checks were introduced.

One is to prevent crashes in the software, because even though I understand your desire to do whatever the hell you want, there are limits where the internal logic stops working (e.g. https://github.com/prusa3d/PrusaSlicer/issues/2715).

Second reason is to stop the user from entering values for which the internal logic stops making sense, even though it does not crash. You can read an outline of flow calculation [here](https://manual.slic3r.org/advanced/flow-math). Of course you are an expert, you don't make mistakes and therefore don't need any checks on any config values. But not everyone is like that (although many think so).

> whoever made this change needs a kick in the seat.
fix this.

Either voice your opinion politely or go somewhere else. In this case you're insulting the lead PrusaSlicer developer @bubnikv and reading you questioning his expertise actually seems kinda funny. We are open to discussion and can make the limit more tolerant if we are convinced it is ok to do. Also, we're not going to obey your direct commands, there is no need to write them. Thanks.

screwyluie, 2019-08-22:
that was me being polite, if you don't like it... well whatever.

> One is to prevent crashes in the software

this is from their default profile. This is something that has worked perfectly fine for every single version of slicer to date. This is not an error and it doesn't crash anything.

>Second reason is to stop the user from entering values for which the internal logic stops making sense

that should be a warning, not a full stop. If it were a warning this would just be annoying. But it's not, it completely stops the program from working.

Also you should be able to turn it off.

>Of course you are an expert, you don't make mistakes and therefore don't need any checks on any config values

ah yes, as expected, tell me to be nice but you can be as passive aggressive as you want... how typical.

tkircher, 2019-08-23:
@Schrunch, you've definitely failed to understand the underlying physical mechanism at work here. Like bubnikv said, if you have a 0.15mm nozzle with properly calibrated extrusion, you will produce a cylinder of material with a diameter of 0.15mm. Which means if your layer heights are 0.15mm, your cross section will consist of barely-touching circles, not squished layers. If you are relying on your extrusion multiplier being set incorrectly in order to extrude enough filament, it means you are using the wrong nozzle size.

Schrunch, 2019-08-23:
The 0.15mm round extrusion theory is good when there is noting under a
nozzle that dont move and the extrusion is just falling down. Also at this
position the output filament diameter is less influenced by the pressure and
acceleration made in the nozzle by the rate of the input filament (that is
habitually a lot larger then the output extrusion). When the nozzle is
moving  over something (or the table under depending of the printer), the
output filament (extrusion) have to bend 90 degrees (we still supposed a
0.15mm dia.nozzle, with a height of 0.15mm). At this moment its upper part
(just under the nozzle) has to do a square bend but his lower part do a
round  bend. Also there is a velocity made to have the filament going out of
the nozzle and it is in relation with the moving speed of the nozzle
depending of the rate put into gcode by the slicer and also influenced by
the software and hardware of the printer.
 
But in any case the extrusion could be perfectly round after being laid
despite any adjustment made because the extrusion his in a melting stage and
his influenced by the velocity that vary form up to down while the extrusion
is being laid (compression modulus on top is different then the stretch
modulus at the bottom and make some delamination that induce deformations).
Also the down velocity force the filament to stick to a plate or to melt
with the printing material under.  If you try this with a rigid steel wire
it will simply broke because of an impossible square bent. The brake will
appears to the round under because the travel is much larger there. If
someone want to go further
 
I understand well that someone put a restriction of bending a filament so
tight but maybe I want to print rows of extrusion just aside one to another
to make some nice looking on the last layer of one of my creation or art
stuff. Is it also possible that I want to print a bridge of a linear rope of
0.15mm between two  balls for example, I dont see the difficulty of
printing those but I do understand that it is impossible to figure all the
case into the slicer code or logic. It is why I am asking to have a warning
instead of a fatal error. If I do something  stupidly wrong, I will be the
one who have to repair my printer!
 
NB: I know that I could draw a 0.16mm rope  or line aside one to the other
(lot of work!), and  put 0.151mm width and play with a negative flow or put
a negative overlap ( I am surprised that this one is available but please
live it there, may be with a warning)
 
I cant call me an expert, I am a creative person who want to explore and do
things I want and do sometimes something that is not expected to be printed
by a slicer. I have a strong logic but there always limits and have to do
trials and errors and  also some mistakes (that make me better!)
Expert is only the name of a level into the Prusa slicer and it I dont know
a better way to call it (may be a  trial and error mode could be add)
 
Note to Thomas Kircher:  accusing someone to definitely failed to understand
something is a personal attack that add absolutely no pertinent information
to the discussion.

 
<https://github.com/prusa3d/PrusaSlicer/issues/2790?email_source=notificatio
ns&email_token=AHVXKKLQOQB7TQ2US3SWXI3QF5IHXA5CNFSM4IN5CBVKYY3PNVWWK3TUL52HS
4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD467UOY#issuecomment-524155451> ,
5IHXANCNFSM4IN5CBVA> .
<https://github.com/notifications/beacon/AHVXKKOZFOHYXGYRCUTTXFTQF5IHXA5CNFS
M4IN5CBVKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD467UOY
.gif>

closed 2026-09-29 by prusaslicer-bot[bot]