prusa3d/PrusaSlicer #2791: Mistake in Polish translation Settings/General/Firmware/Supports remaining time

porlock, 2019-08-20:
### Version
2.0.0+ win 65

In tab Printer Settings/General/Firmware

Text "Supports remaining time" was translated to Polish as "pozostały czas podpór" in this case support ("podpora" in PL) does not means a extra plastic to lift overhangs, it this context to say that this feature is supported you should use word "Wspiera" , so the final translation should be "Wspiera pozostały czas wydruku" just analogical like you do for "stealth mode" text translation in the same tab.

bubnikv, 2019-08-21:
passed to the content guys

bubnikv, 2019-08-22:
our external Polish translator will improve this translation for the next release. 

Thanks for heads up. Closing.

closed 2019-08-22 by bubnikv