prusa3d/PrusaSlicer #2792: M109 command being added at start of GCode

cat007mine, 2019-08-21:
### Version
2.0.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3

### Behavior
At the start of every print, after the custom start printer GCode (which has the printer do a purge line) the printer the run the filament GCode (M900 K200 ; Filament gcode) and then the printer will pause without lifting so the nozzle is sitting in one spot and oozing.
Checking GCode there is a line:

M109 S210 ; set temperature and wait for it to be reached

Where is this M109 coming from? The nozzle is already at print temperature, hence the purge line. All this does is have the printer pause for about 10-15 seconds while it thinks about reheating the nozzle, that's already at temp, then leaves a big blob at the end of the purge line and sometimes will drag it over to the start of the print, which sometimes causes a failure.

Any ideas?

Cheers
Hunt

Jebtrix, 2019-08-21:
It would be helpful if you attached an Exported Config to review.

cat007mine, 2019-08-21:
Sorry - here find attached.

[PLA0.16mm.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/3523435/PLA0.16mm.gcode.zip)

Jebtrix, 2019-08-21:
The short answer is because you didn't try setting extruder temp with either M104; Set temp or M109 set temp and wait in your Printer > Startup G-Code. This triggers PrusaSlicer to automatically add it for you, it's way (explained in its tooltip).

So if you would delete all custom g-code from Printer and Filament you get this (Trimmed for brevity):

```
...
M107
M190 S60 ; set bed temperature and wait for it to be reached
M104 S215 ; set temperature
;Printer start g-code BEGIN
;Printer start g-code END
M109 S215 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;Filament-specific start g-code BEGIN
;Filament-specific start g-code END
...
```

If you just add a extruder temp command, either M104 or M109 it will detect that and not help with extruder temp. NOTE: It didn't detect a bed temp command so it will add that for you.

```
...
M107
M190 S60 ; set bed temperature and wait for it to be reached
;Printer start g-code BEGIN
M104 S215 ; set extruder temp <------
;Printer start g-code END
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;Filament-specific start g-code BEGIN
;Filament-specific start g-code END
...
```

If you just add an bed temp command, either M190 or M140 it will detect that and not help with bed temp. NOTE: It didn't detect a extruder temp command so it will add that for you.

```
...
M107
M104 S215 ; set temperature
;Printer start g-code BEGIN
M190 S60 ; set bed temperature and wait for it to be reached <------
;Printer start g-code END
M109 S215 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;Filament-specific start gcode BEGIN
;Filament-specific start gcode END
...
```

Now if you added a extruder and bed temp command it won't try to add anything:

```
...
M107
;Printer start g-code BEGIN
M104 S215 ; set extruder temp <------
M140 S60 ; set bed temp <------
;Printer start g-code END
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;Filament-specific start g-code BEGIN
;Filament-specific start g-code END
...
```
 
 My recommendation is goto Configuration > Configuration Wizard and add whichever Prusa model yours is closest to. You can use its g-code sections as a reference. Merging yours I would do something like this in Printer > Startup G-Code. 

```
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28; Home
G29; BLTouch mesh generation 
M190 S[first_layer_bed_temperature] ; wait for bed temp
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ;Reset E position
G1 Z0.4 ;Raise Z and start printing

```

lukasmatena, 2020-01-14:
Can this issue be closed then?

cat007mine, 2020-01-14:
Yes please close it.

Thanks

closed 2020-01-14 by lukasmatena

cosworth99, 2023-02-19:
This is not a solution. 

If you enter a print temp in Filament Settings, it should populate **_before_** the start g-code. The wait causes oozing. You absolutely should not have to enter a custom g-code for every temperature change you want to make. This is bad design. Cura puts the M109 command before start g-code and doesn't ooze. 

It still does this in current Prusaslicer releases. Example from today:

`M107
M190 S80 ; set bed temperature and wait for it to be reached
M104 S240 ; set temperature
;TYPE:Custom
G28 ;Home
M420 S;			Enable ABL using saved Mesh and Fade Height

G92 E0 			;Reset Extruder
G1 Z2.0 F3000 		;Move Z Axis up
G1 X10 Y20 Z0.28 F5000.0 	;Move to start position
G1 X10 Y300.0 Z0.28 F3000 E23 	;Draw the first line
G1 X11 Y300.0 Z0.28 F5000.0 	;Move to side a little
G1 X11 Y20.0 Z0.28 F3000 E46 	;Draw the second line
G1 Z5.0 F3000 		;Move Z Axis up
M117 Printing...`

Same file, same start gcode from Cura:

`M140 S110
M105
M190 S110
M104 S260
M105
M109 S260
M82 ;absolute extrusion mode

G28 ;Home
M420 S;			Enable ABL using saved Mesh and Fade Height

G92 E0 			;Reset Extruder
G1 Z2.0 F3000 		;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 	;Move to start position
G1 X10.1 Y300.0 Z0.28 F3000 E23 	;Draw the first line
G1 X10.8 Y300.0 Z0.28 F5000.0 	;Move to side a little
G1 X10.8 Y20.0 Z0.28 F3000 E46 	;Draw the second line
G92 E0 			;Reset Extruder
G1 Z15.0 Y40 F2000 		;Move Z Axis up
M117 Printing...`

Again, you absolutely should not have to enter a custom g-code for changing temperature settings when there is a menu setting in Filament Settings already. That's the point of the menu setting - to avoid having to write custom g-code every time. There are a few UX niggles with this software.

zalexzperez, 2023-02-21:
This action is messing with my Klipper START_PRINT macro.

All I want/need to have for my Start-Gcode is:
`START_PRINT EXTRUDER_TEMP={first_layer_temperature[0]} BED_TEMP={first_layer_bed_temperature[0]}`

Since I want to be able to start printing even if the current extruder temp is higher than the defined extruder temp, the undesired inserted gcode by Prusaslicer at the end of the start-gcode section is making my printer sit still until it cools down (M109) right after my purge line is printed.

opatut, 2023-02-22:
Until this is implemented (if ever), you can use this python script as a [custom post-processor](https://help.prusa3d.com/article/post-processing-scripts_283913):

```python
"""
A post-processing script for PrusaSlicer to remove M104 and M190 commands 
before the custom start macro. Useful if you implement your preheating in 
a macro, e.g. using Klipper, and don't have any preheat commands in your
custom start G-Code in PrusaSlicer anymore.
"""

#!/usr/bin/env python3
import sys

# Adjust these if you use a different macro.
START_PRINT_MACROS = (
    "START_PRINT",
    "PRINT_START",
)

def main(filename):
    result = ""

    with open(filename, "rt", encoding="utf-8") as file:
        found_start = False
        for line in file:
            if not found_start:
                if any(line.startswith(s + " ") for s in START_PRINT_MACROS):
                    found_start = True

                if line.startswith("M190 ") or line.startswith("M104 "):
                    continue

            result += line

    with open(filename, "wt", encoding="utf-8") as file:
        file.write(result)

main(sys.argv[1])
```

Check for yourself it that works properly in your case, I don't guarantuee anything ;)

helmut1337, 2023-04-04:
So this won't be fixed? Issue still exists with Klipper.

JohnOCFII, 2023-04-04:
> So this won't be fixed? Issue still exists with Klipper.

The new alpha 6 works for me.  I was able to remove my dummy commands after selecting firmware type of `Klipper.`.  Are you sure you unchecked this box?

[image]

helmut1337, 2023-04-04:
Thank you! It is working now. Did not have the latest version and thought it wont be fixed because there was nothing labled in here :)