prusa3d/PrusaSlicer #2793: Suggestion - move min_fan_speed to Printer Settings

paulr4, 2019-08-21:
### Version
2.0.0

### Operating system type + version
Windows

### 3D printer brand / version + firmware version (if known)
Marlin 1.1.9 bugfix

### Behavior
This a suggestion for a change to the way fan speeds are handled.

Currently, under Filament Settings > Cooling > Fan Settings, there is a min_fan_speed parameter. Its tooltip reads: 'the minimum PWM setting your fan needs to work'.

This is a feature of the printer, not of the filament, so wouldn't a better place for this be under Printer Settings? Same applies for max_fan_speed. The filament settings could then define a fan speed appropriate for the filament, relative to the upper and lower limits set for the printer.

haarp, 2019-08-27:
The tooltip is wrong. This is actually the fan speed used throughout the print, unless a higher fan speed is needed due to short layer time (up to Max) or due to bridging (bridges fan speed).

rtyr, 2021-02-03:
It should be left in the filament settings. We may improve the tooltip. Closing.

closed 2021-02-03 by rtyr