prusa3d/PrusaSlicer #2794: Feature request: Nonplanar layers

mdjurfeldt, 2019-08-21:
It would be wonderful if it would be possible for you to merge the following Slic3r modification (available on github):

https://tams.informatik.uni-hamburg.de/research/3d-printing/nonplanar_printing/index.php

Venthe, 2019-08-22:
Direct link to Github branch: https://github.com/Zip-o-mat/Slic3r/tree/nonplanar
Released under GNU Affero General Public License v3.0; Same as regular Slic3r

draeath, 2019-08-22:
See also #2704

haarp, 2019-08-27:
This is a super promising feature that would make Prusaslicer stand out quite a lot.

One of the challenges here will be how to handle collision detection. Prusaslicer already does this for sequential printing, but it is quite rudimentary and not adequate. Having the PINDA hang next to the nozzle doesn't help either.

TargoDove, 2019-09-06:
With the current printers it probably wouldn't be very useful/implementable with just a software update. It would also require some clever changes to the current form of the Mk3 and Mk3s. But it would be a nice feature/ability to have for a new printer (say the MK4?). It has been around since 2016 and with all the developments since then it is most certainly possible to implement.

It doesn't have to be perfected before being added to a Prusa Printer. Thats the whole point of releasing new printers. None of them are perfect. I think a lot of us would love to have this feature if it isn't more capable than what was demoed in this video (https://youtu.be/gmePlcU0TRw).

There are and will most certainly be issues and difficulties but that is typically the case with all new features.

alexwhittemore, 2019-09-06:
I mean, certainly Prusa Research doesn't have much incentive to develop slicer features inapplicable to their own printers, but FWIW, PrusaSlicer is definitely used outside the scope of the Prusa i3.

draeath, 2019-09-06:
Even with the small clearance available due to PINDA and part cooling fan shroud this can be useful, just not to the degree everyone hopes.

jurriaan, 2021-03-08:
There's some interesting development going on here: https://xyzdims.com/2021/03/03/3d-printing-90-overhangs-without-support-structure-with-non-planar-slicing-on-3-axis-printer/

Tupson444, 2023-04-03:
With the new Prusa MK4, by installing a longer nozzle (Volcano, using adapter), and printing PETG that doesn't require cooling, since there is no PINDA, it would become capable for printing non-planar.

cfontes, 2023-04-04:
There are nozzles now, with it you can get a nice clearance.
[image]

Tupson444, 2023-04-04:
Yeah even with regular nozzle it would be possible to print low depth non-planar, and because MK4 doesn't require Z offset adjustment, it's easy to install longer nozzles to be able to print deeper non-planar "layers". Fan duct could be modified to blow downwards for a longer nozzle.

In any case, non-planar printing would be very useful (for the looks, structural strength and printability of overhangs) and there is no hardware limit in new Prusa machines (like there was with MK3 using PINDA that would collide with the print), so Prusa team should have direct interest in developing this feature (anyone with MK4 or XL could use it, or anyone else who modifies their machine).

cfontes, 2023-04-05:
Hopefully. 
I don't get why this one is ignored by all slicers. 
Would be interesting to get a glimpse into why. 
Of course this is open source they don't have to do it anything, but this looks like such a "best bang for yout buck" feature... 
It pays for itself in branding with Youtube/online content alone, imagine the headlines, for at least a couple weeks it would be all the rage.

I mean from a product perspective one dev on this makes so much sense at least from our poingant outside view, nevertheless there is no one among all slicers (the big ones) working on anything even close to an Alpha public release, there are scripts and other hacky things but proper integration? Zero.

Maybe it is too specific and no one understands it properlly to write the code? Or just hard to work reliable?

it's a weird one.

mpaw, 2023-05-03:
> Of course this is open source they don't have to do it anything

@cfontes I don't think it's as simple as copying and pasting into the PrusaSlicer source code. There would be a significant time investment to make sure everything works properly and doesn't break any existing functionality. Take a look at how long it has taken to get organic supports working properly, and that code was also open source and written by [someone else](https://github.com/ThomasRahm/CuraEngine/).

> Maybe it is too specific and no one understands it properlly to write the code?

Give it a try and put in a pull request 😃

probonopd, 2026-09-18:
.