prusa3d/PrusaSlicer #2795: Adjust flow of the "second" layer in spiral vase mode

wavexx, 2019-08-21:
### Version
2.1.0-alpha0 (and earlier)

### Behavior
In spiral vase mode, just after the bottom layer is filled, the actual spiral starts using a constant flow and rate of ascent. Since the extruder starts spiraling effectively *at* the bottom layer, the first segments of the spiral are squished, resulting in overextrusion.

This is generally not a problem, but when multiple bottom layers are used, the overextrusion leaves a noticeable bulge higher up in the print, which usually resolves after 3/4 of layer height.

Maybe the the extrusion width could be adjusted for the first segments in order to overextrude less.

ravingDIODE, 2020-04-20:
I have the same issue in all the PrusaSlicer versions I have used so far this year. It is really annoying having this bulge after the 3/4 bottom layers are done printing and the spiral perimeter starts.

Is there a workaround?

So far using Cura, is slightly better and using IdeaMaker, it is gone, but I am struggling to get my IdeaMaker profiles set up nicely.

bubnikv, 2020-12-09:
Linear tapering of extrusion rate at the 1st spiral vase layer implemented with 298b730589371d39fe4a85851a9c72f87fc22d8e

closed 2020-12-09 by bubnikv

bubnikv, 2020-12-09:
Follow-up with 298b730
Linear tapering could only be implemented for relative extruder distances.
For absolute extruder distances, all the G1 Exxx in all spiral vase layers
would have to be adjusted, which is much more complex and risky to do now
before release.