prusa3d/PrusaSlicer #2796: "only infill where needed" sometimes works, usually doesn't

sdotjo, 2019-08-21:
### Operating system type + version
_MacOS Sierra 10.12.6
PrusaSlicer
Version:   2.1.0-alpha1+   also occurred on 2.0.0+
Build:     PrusaSlicer-2.1.0-alpha1+-201908091117

### Behavior

When I choose "only infill where needed" it sometimes generates support as expected, sometimes not. Even going back to exact settings that generated the correct support before, with no change of orientation of the part. 

Steps I take: 
1)choose "only infill where needed". See other relevant settings in screenshot
Sometimes this generates as expected (see screenshot "A"), usually it doesn't (see screenshot "B". 
2)Change a setting, such as infill %. For example from 50% to 30%. Sometimes it generates as expected, usually not. 
3)Change back from 30% to 50%. Sometimes it generates as expected, usually not. 

I've uploaded the .3MF object file.
Here are the screenshots: 

Expected Results: Screenshot A
[image]

More often than not: 
Actual Results: Screenshot B
[image]

Settings:
[image]

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Sierra Version 10.12.6 (Build 16G29)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 INTEL-10.25.17
Vendor:       Intel Inc.
Renderer:     Intel(R) Iris(TM) Graphics 650
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod

Jebtrix, 2019-08-22:
I don't see an attached 3MF project file. Github doesn't know its a zip and will refuse. Just add .zip to the filename.

sdotjo, 2019-08-22:
Here you go:
[Cut4.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3528319/Cut4.3mf.zip)

sdotjo, 2019-08-22:
And if that doesn't work: 
[Cut4.zip](https://github.com/prusa3d/PrusaSlicer/files/3528320/Cut4.zip)

Jebtrix, 2019-08-22:
The file has nozzle at 12mm not 8mm like you have stated. Regardless I could not duplicate this on non MacOS versions. Tried with 2.0 and 2.1 alpha1:

* **Win7 Pro 64bit** - Crashes 100% every time on slicing. I wasn't able to narrow it down to a specific setting. Mesh is clean.

  Getting event log error code of 0xc0000409 - STATUS_STACK_BUFFER_OVERRUN:
  ```
    Faulting application name: prusa-slicer.exe, version: 2.1.0.0, time stamp: 0x5d4d5821
    Faulting module name: PrusaSlicer.dll, version: 0.0.0.0, time stamp: 0x5d4d5815
    Exception code: 0xc0000409
    Fault offset: 0x00000000004de100
  ```

* **Linux-Manjaro (Stable Branch) x86_64 4.19.66-kernel**- No crashes. I get consistent slicing results.

[image]

sdotjo, 2019-08-22:
Ok. How is it on MacOS?

Also, the screenshot you sent is of when it is not working, correct? I want it to look like my Screenshot A, not Screenshot B. 

And yes, should have clarified. It's a 8mm actual nozzle, 12mm settings nozzle. The extra is necessary to boost clay flow.

Jebtrix, 2019-08-22:
I said that Linux version was consistent, not correct or incorrect. I know you wanted results like your Figure A. Try using Print Settings > Bridging Angle of 49degrees instead of 0 (auto): 

[image]

I have never done clay so you will have to play with bridging angle but you should at least get consistent results, acceptable or not.

Jebtrix, 2019-08-22:
I haven't used Infill Where Needed setting much myself. Its a safe bet there is a problem with auto bridging angle, consistent of not. I just try to save devs' time reporting all angles of a problem.

sdotjo, 2019-08-22:
I see. Great. Thanks a lot for your help. I will give Bridging Angle of 49degrees a try.

Jebtrix, 2019-08-22:
> Ok. How is it on MacOS?
Sorry don't own a Mac. There is a problem regardless. Report back if it at least gets consistent with manual angle values just to be thorough on this issue.

sdotjo, 2019-08-22:
So, Bridging Angle at 49 deg did something, but it's still inconsistent. When I changed it to 49, I got expected results, but then changed the fill from 40% to 50% and it went back to little/ weird support added (like in screenshot B). Playing a round with changing both Bridging Angle and fill % (slight changes like doing 46 bridging angle, or +/- 10% fill change) continued to give inconsistent results. Though it seems that it resulted in more instances of behavior like in Screenshot A, than before changing Bridging Angle.

closed 2026-09-29 by prusaslicer-bot[bot]