prusa3d/PrusaSlicer #2797: Infill/perimeter overlap doesn't work correctly if values are set per object.

rtyr, 2019-08-22:
### Version
PrusaSlicer 2.1.0-alpha1

### Operating system type + version
Win10_64

### Behavior

Infill/perimeter overlap doesn't work correctly if values are set per object.

* _Steps needed to reproduce the problem:

Open attached 3MF project file.

* _Expected Results_

Cubes with 25%/50%/100% infill/perimeters overlap.

* _Actual Results_

All cubes have 25% perimeters overlap.

[3mf_overlap.zip](https://github.com/prusa3d/PrusaSlicer/files/3529032/3mf_overlap.zip)

[image]

Jebtrix, 2019-08-22:
Yup. I see a problem of _multiple objects_ **with** overlap object setting modifiers.

* 1 object on plate - Global or Overlap Object Modifier works fine
* 2 objects on plate, 1 object with Object Modifer - Global change will work on unmodded object BUT if (global = modifier value) then modified object changes properly.

rtyr, 2020-09-29:
It seems fine in 2.2.0.

closed 2020-09-29 by rtyr