prusa3d/PrusaSlicer #2798: Pause for Material Change

Rice-Crispy, 2019-08-22:
### Version
_Version of PrusaSlicer used goes here_2.0.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 10 x64
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ 
Prusa MK3 stock
Firmware installed - prusa3d_fw_MK3S_3_7_2_2363.hex

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Printing an ID tag part, basic square shape (32x32mm) that is a total of 4mm thick.  I added text 1mm deep to both sides and a through hole.  My idea was to print the first 1mm in silver, 2mm in orange, and then 1mm in silver.  After slicing the part in PrusaSlicer 2.0.0 I used the vertical bar (to the right of the graphical area) and dragged up to the 1.00mm level then clicked the "+" button.  The graphics area shows the last layer of text on the bottom surface, and after this layer I want to swap colors.  It worked flawlessly.  However, I also added a material change at the 3.00, where the graphics area shows the layer just before my text begins on the top of the part. So ideally I've printed 1mm of silver, 2mm of orange, and then change materials back to silver.  But once the part was complete I noticed that at least one layer (probably only one layer) of extra material was printed and I don't have the color contrast I wanted to see in the text.

_Is this a new feature request?_No
[image]
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[xyz SMALL DIAMOND.zip](https://github.com/prusa3d/PrusaSlicer/files/3530913/xyz.SMALL.DIAMOND.zip)

YuSanka, 2019-08-22:
As was described in comments for #2393 
> Every slider tick(position) corresponds to a printing layer on a 3D scene. Clicking on "+" is used for a color change on a SELECTED layer.
Furthermore, each layer is colored in Preview to be as much as possible obvious ;)

closed 2019-08-22 by YuSanka

Rice-Crispy, 2019-08-22:
Thank you for the prompt response.  I have viewed my original part and it seems like I assumed the material change occurred after the shown layer (which is how my Markforged slicer/printer works).  It is now obvious to me that the material change occurs before the layer being displayed in the graphics area.  So my “+” was off by 1 layer twice on the same part.  I have since adjusted my thinking and slicer options and successfully printed the part.

Michael Rice
IE Assembly Design
Henderson Plant
Chassis & Safety Division / Hydraulic Braking Systems Business Unit

Continental Automotive Systems Inc.
1 Quality Way Fletcher, NC 28732
Tel: +1 (828) 654-2344
Mobile: +1 (828) 606-5454
E-Mail:michael.2.rice@continental-corporation.com