prusa3d/PrusaSlicer #2800: Printer fails to print file with special characters (^) in name

jokohoko, 2019-08-22:
### Version
Prusaslicer-v2.1.0-alpha

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Original Prusa MK2.5S and MK3S

### Behavior

This bug annoys me mostly because Solidworks for excample often saves files with ^ in the filename when working in assemblies: eg: Part3^brandblushouder.gcode .
I have not gone trough a list of characters to make an overview but i guess a solution would either be for this specific character or for all special characters at once.
The printer is able to display the name correctly. But he states the file is incomplete upon starting the print. (I am certain the file is complete and after changing the name it prints fine).

I propose the software would replace these characters with working ones like _ or just removes them.

I also reported this to the firmware issues page since another solution is to make the printer capable of printing these files.

[Part3^brandblushouder.zip](https://github.com/prusa3d/PrusaSlicer/files/3531725/Part3.brandblushouder.zip)

Jebtrix, 2019-08-22:
I'm assuming your using SD card to print from? The Marlin firmware its based on only uses 8.3 filenames for file commands besides the hacky M33:Get Long Path. You can try one trick as a workaround. Im not at printer to try but make your part names longer than 6 characters in Solidworks. 

Your 8.3 shortname for that example is:
`PART3^~1.3MF Part3^brandblushouder.3mf`

If you just had a longer part name it would be:
`PART3X~1.3MF Part3x^brandblushouder.3mf`

Easiest way to see shortnames is do a `dir /x` in windows console window.

jokohoko, 2019-08-23:
Yes from SD card.
Indeed the longer filename solves the issue on the printers side. But of course just changing the name of the file also solves the issue. 
If possible i would propose a simple character replacement in the save code so that ^ would just become _ 
Solidworks is not the only software with this convention, i have similar issues with files from Onshape.

Jebtrix, 2019-08-23:
I understand and agree. If you do a bunch of iterate testing its just easier to change in solidworks once, then everytime you generate a gcode file.

jokohoko, 2019-08-23:
The situation is that I print for our students at the university of Antwerp. So on average I get like 8 of these files per month. And changing the standard on all pc's is impossible.

A fix in the slicer would be ideal and could save other users with comparable issues the confusion.

I just tried to find the save code. But seem unable to find the location where to search in the code. My basic C understanding could help suggesting a solution.

-------- Original Message --------
From:Jebtrix
Sent:Fri, 23 Aug 2019 19:22:55 +0200
To:prusa3d/PrusaSlicer
Cc:jokohoko ,Author
Subject:Re: [prusa3d/PrusaSlicer] Printer fails to print file with special characters (^) in name (#2800)

I understand and agree. If you do a bunch of iterate testing its just easier to change in solidworks once, then everytime you generate a gcode file.

Jebtrix, 2019-08-23:
I don't know here my head was. Just go into Print Settings > Output options > Output file > Output filename format and change it. Just get {input_filename_base} out of the beginning and do a placeholder combo that is always 6+ characters or "hardcode" something in the beginning 6+.

bubnikv, 2019-08-25:
The G-code output file names are already being processed to remove the localized non-7bit characters
see void Plater::export_gcode(), call to Slic3r::fold_utf8_to_ascii()

We may add some more path mangling there. What characters should be avoided and what they should be replaced with?

jokohoko, 2019-08-25:
I did some research and testing: 
In Ascii codes valid characters include: ~, ^, @, &, *, and more.
Some are not allowed by windows: `\/:*?"<>| ` so these get a notification that the filename is invalid upon saving (I cannot test on a Mac or Linux to see what they allow).

I made test files including the list of output characters from this [sourcecode](https://github.com/prusa3d/PrusaSlicer/blob/56dfb9f8b0cf99cd8bc62f00c82b65c3bdc4792e/src/slic3r/Utils/ASCIIFolding.cpp). 
Thereby i found only 2 characters that have bugs:

The ^ character as mentioned makes the printer think the file is incomplete and makes it impossible to print the file.
And the ~ character that cannot be visualized on the printer, but works just fine to print (it shows → instead).

So i made a pull request from these changes #2817

Jebtrix, 2019-08-26:
On linux native filesystems **only** forward slash / is not allowed in a filename. An error would only occur trying to save gcode file to a NTFS/Fatxx file system

bubnikv, 2019-08-26:
What printer did you test with?

> The ^ character as mentioned makes the printer think the file is incomplete and makes it impossible to print the file.

Is the problem caused by the file name itself, or by entering the file name as a comment somewhere into the G-code?

> And the ~ character that cannot be visualized on the printer, but works just fine to print (it shows → instead).

Does it really need to be replaced then? I would say no.

jokohoko, 2019-08-26:
I will update this evening.
I tried the same tests on my printers at work (3x MK2.5S and 5x MK3S) and all of these work just fine with alle the special characters.

My printer at home (an MK2.5S) cannot print the files with ^ in the name. I tried various firmware's and files. So will re-do the tests later today to finish this off.

bubnikv, 2019-08-27:
Thanks for testing. Maybe we shall fix the printer firmware. Let's see.

bubnikv, 2019-08-27:
I have informed our testers and firmware team to look into it.

DRracer, 2019-08-27:
The "^" was especially tricky and had nothing in common with PrusaSlicer - requests to start printing filenames with this character got blocked away - the printer was assuming the command was something else. We have a fix, we'll do proper testing in the upcoming days. Thanks for your report.

jokohoko, 2019-08-27:
Ok thanks!
Yeah i was quite bafled by this one since out of the 8 prusa printers i have acces to 2 fail to print the file, the 6 others have no problem with it.
I also opened an [issue on the firmware side](https://github.com/prusa3d/Prusa-Firmware/issues/2130) since i was unsure where the fault/solution would be

bubnikv, 2019-08-27:
@jokohoko could you please close this issue then? Thanks.

closed 2019-08-27 by jokohoko