prusa3d/PrusaSlicer #2801: Feature Request: Option to combine results of 'support from build plate only' and 'support enforcers only'

Blargedy, 2019-08-23:
### Version
2.1.0-alpha1

### Operating system type + version
Windows 10 64-bit

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3s MMU2s

### Behavior

I would like to be able to add a couple spots of support in addition to the supports generated when I select "from build plate". Currently I'd have to either do support everywhere with support blockers, or support enforcers everywhere I want supports, both of which are tedious for the two parts I want extra.  

This is the part I want to print [image]

These are the two spots supported with support enforcers [image]

This unfortunately leaves the bottom unsupported [image]

It is quite complex so I don't want to use 6 different support enforcers or 4-6 support blockers with support all selected. 

I would need the support enforcer generated supports plus the build plate only supports [image]

[project file for reference](https://github.com/prusa3d/PrusaSlicer/files/3532374/Selector_Ema_v9.zip)

Jebtrix, 2019-08-23:
It would be a nice option for flexibility to combine Enforcers and From Build Plate support. 

As for your particular model just using Everywhere with **5%** Overhang threshold adjustments seems to get what your looking for:

[image]

Jebtrix, 2019-08-23:
Actually even Build Plate Only with low Overhang Threshold gets you same place. I guess to match exactly your posted pictures one support blocker would have to be used.

[image]

JohnnyDeer, 2019-11-07:
Hi, in the latest version of PrusaSlicer is possible to combine support on build plate only with support enforcers.

MikePfaff, 2020-02-02:
I support this feature request. I am trying to print 1:1 models of human brains (based on MRI scans) and "build plate only" generates most of the supports needed for a successful print, but in 2-3 places the auto-generated supports are not sufficient and the resulting print has ugly strings/blobs in these places.

Thus I would like to be able to add manual supports - in addition to the "build plate only" generated ones - in these specific places. Having to manually reproduce all the "build plate only" supports (either by using "everywhere - support blockers" or "support enforcers only") would be extremely tedious, as the 3D model is quite complex.

I might start playing around with the suggested "overhang threshold" workaround, but this seems rather time and filament intensive for each try (69h, 600g), although I might be able to make testing easier by just cutting up the model and printing the affected areas only.

A "build plate + support enforcers" option would probably solve all these issues for me. Maybe it could also make sense to think about a "build plate - support blockers" option for other peoples use cases?

MikePfaff, 2020-02-02:
After looking at src/libslic3r/SupportMaterial.cpp and trying in PrusaSlicer 2.1.1, i can confirm that this already works. Just add support enforcers to your model, select "Support on build plate only" and Slice. 

Thus suggesting that the wording in the GUI should be changed to say "Support on build plate" (without the "only"), so that it is clear to the user that support enforcers will still be added. The word "only" kind of implies to the user that nothing else gets added to the build plate supports.

Actually, this "only" word is what made me look for a fix / workaround - and discovering this open feature request - without even trying. I just assumed from the GUI that i could not add anything to "build plate **only**" supports.

Blargedy, 2020-02-09:
@MikePfaff I am unable to replicate your results of getting both enforced and build plate supports. I am on PrusaSlicer 2.1.1. Attached is the 3mf file for a part I am currently printing, used to test your claim. 
[Cutlery_OrganiserSupportTest.zip](https://github.com/prusa3d/PrusaSlicer/files/4176534/Cutlery_OrganiserSupportTest.zip)
What the support enforcers do however do is add supports that are both from the build plate up AND intersecting with the support enforcer. I would like a feature that combines with logic OR the "Support on build plate only" and "For support enforcers only". In the attached file I would want the supports on the two sides as well as supports between the top edge and bottom wall. 
[image]

zalexzperez, 2020-02-16:
Definitely not working on PS 2.2.0 Alpha-4
You're forced to select supports everywhere and start adding many support blockers when only a single support enforcer + touching buildplate would've been perfect.

[image]
[image]

JohnnyDeer, 2020-03-09:
Combination of "support on build plate only" with enforcers are not working ideally right now. We know about it and it is on our to do list.

buzkill420, 2021-02-01:
Bump! Much Needed feature. Very annoying not to have the option and- to me anyways- seems like a large oversight to not include this option to begin with.....

bubnikv, 2021-03-18:
The support generator has been reworked to give the support enforcers precedence over "on build plate only".

[image]
[image]
[image]

bubnikv, 2021-03-18:
[image]

bubnikv, 2021-03-19:
Fixed with 9f09f03228478aa77549efec845717b06e46cbc3

closed 2021-03-19 by bubnikv

nirurin, 2021-03-31:
So this got closed... but I was having this same thought today, and actually found this thread, and then noticed that my prusaslicer was still on 2.2 and wasn't updating to 2.3 (had to install it fresh from the website) and got it all installed and transferred over......

But support enforcers still don't work, or at least they don't override the 'support bed only' option.

Blargedy, 2021-05-22:
@bubnikv I don't know why people keep closing this when it is definitely not implemented. Take any stl, set it to support from buildplate only, then add a support enforcer somewhere and see how it doesn't add them there.

lukasmatena, 2021-05-22:
@nirurin @Blargedy Nobody "keeps closing this", it was only closed once - on 19 Mar 2021, after the issue was fixed with 9f09f03. It should not be surprising that version 2.3 (which was released in January) did not magically fix itself. You'll have to wait for 2.4 release or compile the source yourself. 2.3.1 is a minor patch release which was never supposed to include all the fixes.

I just checked that the 3MF posted in the first post does work in current master (!). If you compiled it yourself and it does not work, we'll be glad if report it (send a 3MF and exact commit so we can reproduce it). The issue will be reopened in that case.

Blargedy, 2021-05-23:
@lukasmatena that's quite the sass you have. I've personally never seen a program with the feature organization as prusa slicer so when it was said the issue is solved and the newest edition doesn't share this solved issue it isn't immediately obvious that you have to be on an unofficial release to see the feature.