prusa3d/PrusaSlicer #2802: Feature request: retraction, extruder and ventilation parameter per layer range

liftbag, 2019-08-23:
### Version
_PrusaSlicer 2.1.0-alpha1+_

### Operating system type + version
_not relevant _

### 3D printer brand / version + firmware version (if known)
_not relevant_

### Behavior
_Since from the 2.1.0 alpha version the layer ranges have finally been introduced, I wonder if it is possible to assign the extruder temperature, ventilation and the entire set of retraction parameters to the layer range_
* _Steps needed to reproduce the problem_
  * _It is not really a problem, but sometimes, and depending on the filament used, it would be possible to improve the final result by varying the parameters of retraction, temperature and ventilation in certain layer ranges.
For example, in the print visible in the image, the holes due to retraction, only appear when the layer height drops to 0.10mm.
In the same way, I would like to be able to comfortably change the temperature and the ventilation in a range of layers where the layer height and print speed are very small, without having to insert a script in the section "Before layer change G-code"._

[repaired_thanos-bust-fdm.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3533496/repaired_thanos-bust-fdm.3mf.zip)
[image]

_Is this a new feature request?_
Yes it is.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

ProfEngr, 2020-01-07:
+1 for including these options. Their absence is one of the few reasons I have to keep Simplify3D around and it's gone wonky with extrusion multiplier settings lately.

rtyr, 2021-02-03:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/3635. Closing this one.

closed 2021-02-03 by rtyr