prusa3d/PrusaSlicer #2803: Strange results when changing perimeters with "edit layers"

jdpurcell, 2019-08-23:
I'm using PrusaSlicer 2.1.0 alpha1. Start by loading any object on the build plate (e.g. a simple cube [(STL)](https://github.com/prusa3d/PrusaSlicer/files/3535000/Cube.zip)).

Slice and make a note of the print time:
[image]

Right click on the object and click "Edit Layers". Right click on the layer editing icon and click "Layers and Perimeters". Change perimeters to "1".
[image]

Re-slice and notice that the print time has gone up significantly, and it is not printing with 1 perimeter as requested on the lower layers:
[image]

I think it's trying to double-print some parts of the object now. Maybe the perimeter setting did take effect, but you can't see because it's also printing a 3-perimeter version of the cube in the same place.

As a workaround, if you add another layer range with default settings, it seems to come out as expected:
[image]

Jebtrix, 2019-08-23:
I can't reproduce that. Can you attach the project file?

jdpurcell, 2019-08-23:
@Jebtrix [Cube-Project.zip](https://github.com/prusa3d/PrusaSlicer/files/3535784/Cube-Project.zip)

Jebtrix, 2019-08-23:
**Enable** Print Settings > Advanced > Other > Clip multi-part objects

Looks like layer modifier is based off multi-material functionality. So that setting applies even though your not using it in that context.

Jebtrix, 2019-08-23:
And to the devs Clip multi-part objects should be automatically checked and enabled using layer modifiers if it does this no? Or maybe disconnect the option to layer edits and its just implied internally?

bubnikv, 2019-08-25:
Lkely the clip multi-part objects should be always active, and we should remove the checkbox from the UI. This was a new addition compared to the upstream slic3r, so we kept the checkbox there for compatibility. I see not much reason for NOT doing the clipping.

Jebtrix, 2019-08-25:
To be honest I thought the same thing. I can't even think of a clever use for it in traditional plasticsm I'd love to hear one though. Then again there is the corner case of 3D printing edibles, chocolates, confections and the like. Only thing I can think of.

jdpurcell, 2019-08-25:
FWIW, although the Prusa printer presets have this on by default, it's actually off by default if you define a custom printer, and only visible in Expert mode. So I could definitely see it causing some confusion.

CapnBry, 2019-09-16:
Another workaround for this is to set the "Start height" of the Range to 0.001. It seems like something is checking the start height for >0 or !=0, which indicates that it is a bug somewhere that's actually being fixed / cleaned up by the multi-object clipping procedure. All other modifier meshes and per-object settings don't generate overlapping gcode with both different settings (even with Clip multi-part objects off).

bubnikv, 2020-03-20:
This issue seems to be a manifestation of a9fc39491e0b37f9a68ac99c372f09121e4c8afd
Indeed, with a9fc39491e0b37f9a68ac99c372f09121e4c8afd applied I am unable to reproduce the issue reported, but I trust you it was there.

Another case of this issue is #2540.

Closing. The fix will be in PrusaSlicer 2.2.0 final.

Also I have enabled "clip multi-part objects" by default.

closed 2020-03-20 by bubnikv