prusa3d/PrusaSlicer #2804: Silent Install Option Hangs

optimiz, 2019-08-23:
### Version
Prusaslicer 2.2.6

### Operating system type + version
Microsoft Windows 10 Enterprise LTSC Version 10.0.17763 Build 17763 (64bit)

### Behavior
* _Describe the problem_

Unable to silently install, internal sub-installer (netfabb.exe? or dpinstx64.exe?) hangs, even when driver integrity enforcement is disabled, so install process never completes

* _Steps needed to reproduce the problem_

1. bcdedit -set loadoptions ddisable_integrity_checks
2. prusa3d_win_2_2_6.exe /sp- /silent /norestart

NOTE: I originally tried without disabling integrity checks and made even less progress.

* _Expected Results_

Silent installation should complete within two minutes, like when run directly.

* _Actual Results_

Installation hangs.

_Is this a new feature request?_

No.

Is there a method to silently deploy Prusaslicer successfully?  I'd prefer to automate this install for the number of units in the labs here.  Thanks for any guidance.

bubnikv, 2019-08-25:
I have passed the issue to the guy responsible for the Windows installer.

bubnikv, 2019-08-27:
You may just pull the binaries from github releases and unpack them with your silent installer. The only difference would be, that the github Windows binaries are not signed. We should improve that.

opsi-archIT-KIT, 2019-12-12:
Installation of PrusaSlicer 2.1.0 (extracted, included in Prusa3D 2.2.8, Inno-Installer) is also not silent because of checking if update exists (probably release on github!?). Since a few days there is version 2.1.1 out.

Assuming that checking for updates is implemented with a "Run"-Section in Inno: then please consider to use the "skipifsilent" option. Look at http://www.jrsoftware.org/ishelp/index.php?topic=runsection

@bubnikv: will try to use the zipped version from here. Thanks for the hint.

opsi-archIT-KIT, 2020-01-23:
Also uninstallation of PrusaSlicer 2.1.0 with inno (un)installer is not silent! Popup window with "Do you really want to uninstall?" and afterwards a useless confirmation window that now it is uninstalled appearing.

PrusaSlicer is not ready for deployment systems although the inno installer essentially is able to do that.

Using now the zipped version for deploying PrusaSlicer.

lukasmatena, 2020-03-04:
This should be solved in new installer that will be used for 2.2.0 final. Let us know if it isn't.

Thanks for reporting. Closing.

optimiz, 2020-03-24:
Thank you for the update; I appreciate the effort that went into this.  I'll run a test install once the COVID-19 shutdowns conclude.

Orava2, 2021-08-25:
Command
    prusa3d_win_2_3_0.exe /sp- /silent /norestart
shows a progress window and opens webpages. Installation will stop if newer version is found from Internet.

The uninstaller will ask "Are you supre you want to completely remove PrusaSlicer and all of its components?".
    "C:\Program Files\Prusa3D\unins000.exe" /sp- /silent /norestart

jake-the-creative, 2022-01-25:
This is still an issue
https://forum.prusaprinters.org/forum/prusaslicer/how-to-silently-install-and-uninstall-prusaslicer/
There appears to be no way to actually silently install prusiaslicer ?

hypercube33, 2022-10-01:
I can help with this if you need it. What people are looking for is to be able to mass-deploy the software to multiple machines using management tools (Group Policy, SCCM, other tools) and when you do that you need to run the installer in what they call "Silent" mode where it doesn't prompt, or ask anything - no popups, nothing. Just installs. Same with the uninstall - if they need to remove it.

I'm familiar with Inno Setup, Nullsoft Scriptable Installer System, etc. and can help but I dont see the ino setup files on the repository otherwise I'd have checked out what it would take to get this working. Typically, out of the box this functionality works, but it sounds like there have been custom setup wizard pages or addons added that do things that break the silent installer flags which is what people are complaining about.

TLDR - Inno Setup by default has some flags to run the installer automatically, no prompts, from a command line that results in the software just being installed - nothing else, no user interface, no errors etc. 

Currently, the downloadable installer just returns errors at the command line when you try standard setup parameters.

HabitualTinkerer, 2023-02-22:
dpinst64.exe is prompting to trust drivers from Prusa Research, which doesn't appear if the installer is running as SYSTEM.

You can work around this problem:
1. Install Prusa Slicer package with drivers manually.
2. Export the "Prusa Research s.r.o." certificate from the Trusted Publishers machine store. I chose to name the file `prusa_cert.cer`.
3. Import the certificate to Trusted Publishers **before** starting the silent install. I'm sure this can be done with certutil but I used PowerShell: `Import-Certificate -FilePath prusa_cert.cer -CertStoreLocation Cert:\LocalMachine\TrustedPublisher`

This should prevent the invisible prompt from appearing when dpinstx64.exe runs.

closed 2026-09-29 by prusaslicer-bot[bot]