prusa3d/PrusaSlicer #2807: Question: Possible to calculate with placeholders

TheLion, 2019-08-24:
Is it possible to calculate with placeholders like [first_layer_bed_temperature] to increase the temperature set for the filament by 10c, in case of different printers or bed surfaces? To prevent creating a filament entry for every printer.

Something like [first_layer_bed_temperature+10]

Tried this syntax, but that failed.

lukasmatena, 2019-08-24:
It is possible. There are two issues  with your attempt:
1. You are using legacy square brackets
2. You are not subscripting a vector variable
Try `{first_layer_bed_temperature[0]+10}`.
See also https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

Jebtrix, 2019-08-24:
> You are not subscripting a vector variable

You realize how alien that is to the average user right?? :smile: That wiki page should explain to users that the array variable stands for which filament to reference.

bubnikv, 2019-08-25:
@Jebtrix 

> You realize how alien that is to the average user right?? 😄 That wiki page should explain to users that the array variable stands for which filament to reference.

Could you edit the wiki page?

There are array variables not only for filaments, but also for extruders, bed contour points etc.

Jebtrix, 2019-08-25:
@bubnikv I actually started to but since github doesn't allow PRs on wiki page edits (which is a shame) I felt it would be a little pretentious. I'd like to do a more end user centric (over explain things i guess :smiley: ) example driven documentation, workarounds etc. Attempt the start of an actual manual maybe. I go down the rabbit holes to understand the nuances but I'm sure I'll miss things being on the end user side of the fence. Plus I don't own all supported hardware types SLA printer etc.. There is the https://pages.github.com github.io functionality that at least brings PR support but it would be another Prusa3D repo. Upside is all Prusa3D having a central github help doc, more flexibility in the content and the PRs alone may be worth it. Have a section people could PR non Prusa printer config examples, etc.

I could duplicate the wiki here on my own github.io to flesh out some ideas. Then you can yea or nay it? I just don't want to waste anyone's time, including my own.

bubnikv, 2019-08-26:
> explain things i guess 😃 ) example driven documentation, workarounds

If you update the wiki page in your fork, we will likely use it.

Our content team is working on PrusaSlicer manual, they will likely
integrate our wiki.

bubnikv, 2020-03-26:
The placeholder language is described here.
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language
Closing.

closed 2020-03-26 by bubnikv