prusa3d/PrusaSlicer #2808: Issue with supports not generated

Yezariael75, 2019-08-24:
### Version
2.1.0 Beta Win64

### Operating system type + version
Windows 10 64bit

### Behavior
I created a little overhang test for support material and the supports are not generated everywhere where needed. Works on 2.0 with same presets and .stl.

#### Project File (.3MF) where problem occurs
attached
[image]
[hips-test.zip](https://github.com/prusa3d/PrusaSlicer/files/3537240/hips-test.zip)

thisiskeithb, 2019-08-24:
I noticed a similar issue as well despite supports set to "everywhere" and toggling "Don't support bridges" doesn't seem to do anything. I tried with multiple profiles and the issue still persisted.

## 2.1 alpha:
[image]

## 2.1 beta:
[image]

## 3MF/Project file:
[asa-test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3537299/asa-test.3mf.zip)

smitty97, 2019-08-24:
Same issue here 2.0.0 vs 2.1.0
2.0.0 (correct):
[image]

2.1.0b:
[image]

support blocker did not work in 2.1.0b either.

3mf file:
[pinkys revenge podv4.zip](https://github.com/prusa3d/PrusaSlicer/files/3537326/pinkys.revenge.podv4.zip)

Jebtrix, 2019-08-24:
Yeah supports is just plain broke. You can nudge it to try rotating object but its even starting supports in mid air.

[image]

bubnikv, 2019-08-25:
That is pretty bad. I guess this issue was created with a seemingly harmless refactoring dc3a0a0ab34af1ac0e0139cbde2a75f820e17f1f . I will try to patch it and create a beta2 soon.

bubnikv, 2019-08-25:
Fixed with ceae7b4889342daa6babf0de244e0f9175ef65a3
The beta2 will be released after a quick testing.

Jebtrix, 2019-08-25:
@bubnikv Went thru some of my catalog of support centric projects. Looking good.

cosmith71, 2019-08-25:
Will there be an installer?  I generally don't have much luck trying to compile.  :(

bubnikv, 2019-08-25:
The beta2 has been published.
Closing.

closed 2019-08-25 by bubnikv