prusa3d/PrusaSlicer #2809: gcode update function

cnlson, 2019-08-24:
### Version
2.0.0+win64
### Operating system type + version
win7 x64
### 3D printer brand / version + firmware version (if known)
Prusa mk3 --> mk3s
### Behavior
_Is this a new feature request?_ YES

I have had my mk3 since December of 2017 I have used prusaslicer, prusa slic3r and prusacontrol
Along the way remaining time has been added, I've updated my start and end gcode to custom settings and what i would really like is a feature refresh my old gcodes to add these new features without requiring that i re-find the old STLs replate everything exactly as I had optimized them to in the last 2 years, reslice them, and resave them. 

currently in prusaslicer there is no import for a gcode file so it is not possible to do even manually but I would hope that it could be done as a batch. All the files on the sd card at once.

Jebtrix, 2019-08-24:
Upgrading your hardware seems like a good time to re-calibrate to me. Personally I wouldn't want any slicer to attempt any custom gcode modification on my behalf. That is just asking for trouble.

> it is not possible to do even manually

How do you come to that conclusion? Set PrusaSlicer to old Printer profile, copy a gcode section, paste into ie Notepad then open new default MK3S profile and do the same. Merging should be trivial, then just paste into MK3S profile and save as new custom. If you want to go gung-ho you can use something like [Winmerge](https://winmerge.org/). Just save old and new gcode text to different files first. 

Comparing the entire actual profile ini files from Prusa in c:/users/<username>/appdata/roaming/prusaslicer/printer is also possible but takes some prep because of how it saves gcode. If you use Notepad++ you can do a Find/Replace with extended search mode and replace `\\n` with `\n` for easier comparing of gcode segments.

cnlson, 2019-08-24:
> > it is not possible to do even manually
> 
> How do you come to that conclusion? 

i'm pretty sure you cannot change to remaining time that way.

Jebtrix, 2019-08-25:
You meant change the final output gcode file.. I misread the _remaining time_ feature is the crux of your request. I digress.

bubnikv, 2019-08-25:
You can likely try to re-slice all models using the PrusaSlicer in command line model while loading the old G-code as a configuration. This will work for single object prints if your G-code file name contains the name of the STL.

Sorry, but we will not implement your request. Current PrusaSlicer could slice 3MFs even with the command line slicer. Your request is for supporting results of a legacy slicer and before we would implement your request it would be needed no more.

closed 2019-08-25 by bubnikv