prusa3d/PrusaSlicer #2810: Support generation problem

tkapler, 2019-08-25:
### Version
Version 2.1.0-beta+win64

### Operating system type + version
Windows 10 (build 18362), 64-bit edition

### 3D printer brand / version + firmware version (if known)
MK3S latest firmware

### Behavior
generated supports are wrong - sometimes missing, sometimes supporting nothing. See image 2.0.0 vs. latest alpha for same object

[image]
[image]

[chest poppy2.zip](https://github.com/prusa3d/PrusaSlicer/files/3537689/chest.poppy2.zip)

bubnikv, 2019-08-25:
https://github.com/prusa3d/PrusaSlicer/issues/2808. Closing this one.

closed 2019-08-25 by bubnikv

Shama71, 2019-08-25:
Hi.
In the screenshots, the same object, but placed on different sides.

Dont support bridges is on
[image]
[image]

Dont support bridges is off
[image]
[image]
[Support.zip](https://github.com/prusa3d/PrusaSlicer/files/3537885/Support.zip)

bubnikv, 2019-08-25:
it is simply broken due to a tiny bug. A fixed beta2  is due in couple of hours.