prusa3d/PrusaSlicer #2812: Clear undo chain

neslekkim, 2019-08-25:
The new undo feature is awesome!

But, when I do new project, then I can still undo, it should be at some point be able to really clear the slicers undobuffer.
(and maybe also clear all settings, instead of the need to close and restart slicer)

Jebtrix, 2019-08-25:
Very rare to have a feature go too awesome :laughing: I'd even go for a message asking/informing if you want to go back to previous project. Could come in handy.

bubnikv, 2020-03-14:
> (and maybe also clear all settings, instead of the need to close and restart slicer)

Just re-select the particular profile. It will be reloaded.

I find it nice to have all the preceding operations on the Undo / Redo buffer including the project loads.

neslekkim, 2020-03-14:
It's like no other tool that does that.. I close slicer and re-open, but it's getting slower and slower at startup.
keeping settings when doing new project is not cool at all.

bubnikv, 2021-09-30:
It will be implemented in PrusaSlicer 2.4.0-alpha3.

closed 2021-09-30 by bubnikv