prusa3d/PrusaSlicer #2813: Error on importing stl

kdsohn, 2019-08-25:
### Version
2.0.0+win64, recent updates

### Operating system type + version
WIN 10 64bit
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa MK3S, not connected

### Behavior
Error on importing a STL object
Start PrusaSlicer APP
Import STL File
An error is diplayed:
[image]
and nothing more happens

_Is this a new feature request?_
no

lukasmatena, 2019-08-25:
If you don't enclose the STL that's failing, there is not much we can do for you. Apart from saying that your STL file is probably corrupted or was incorrectly generated in the first place.

kdsohn, 2019-08-25:
It's the same with all stl files I have. I have attached some now.
[DTCO-NINA.zip](https://github.com/prusa3d/PrusaSlicer/files/3538379/DTCO-NINA.zip)

jokohoko, 2019-08-25:
From what software did you export these files?

(Without being code savvy it will be hard to do modifications. All you can do it to try changing export settings to binary stl files. Since these are a bit stricter they work better.  
Below is more info for the developers if they try to solve this in the code).

It's data looks like:
```
solid DTCO-NINA-Boden
  facet normal -0.00000000 -1.00000000 -0.00000000
    outer loop
      vertex 64.00000000 9.05000000 1.00000000
      vertex 35.00000000 9.05000000 1.00000000
      vertex 64.00000000 9.05000000 0.00000000
    endloop
  endfacet 0
  facet normal 0.00000000 -1.00000000 0.00000000
    outer loop
      vertex 64.00000000 9.05000000 0.00000000
      vertex 35.00000000 9.05000000 1.00000000
      vertex 35.00000000 9.05000000 0.00000000
    endloop
  endfacet 0
```
Where normal STL files never have numbers behind the endfacet lines.
When all numbers are unique the slicer is fine, but with these files about 30 facet's have the same endfacet number. And this crashes the slicer. I tried removing all numbers and then the files loads without a problem.

Jebtrix, 2019-08-26:
On 2.1.0beta2 it just displays a "Loading of a model file failed". Its two quick fscanf format changes in source to take these files. It seems all my usual 3D/CAD/Other Slicers import these files just fine so I guess in the end it doesn't really matter who/what made them.

kdsohn, 2019-08-26:
So message was misleading me. I assumed a software module couldn't be loaded and I did not relate the message to the content of the model.
STL binary work better, but I hadn't this issue with any other software, so it wouldn't be bad to get that solved.

closed 2019-08-26 by kdsohn

bubnikv, 2019-08-26:
The ASCII STL import was made more tolerant to non-standard input format,
namely:
1) Whitespaces are allowed at the end of lines
2) A freeform is allowed after "endloop", "endfacet" and "endsolid", if separated
from the keyword by a whitespace. Some ASCII STL exporters likely add their
annotations in there.

Fixed with 9abef2241d188d0cc0892bc3070d7360749c2702