prusa3d/PrusaSlicer #2814: Delayed manipulation with view

DominikPalo, 2019-08-25:
### Version
PrusaSlicer-2.1.0-beta+-201908231457

### Operating system type + version
System Version: macOS Version 10.14.6 (Build 18G87)

### Behavior
* Start holding the left mouse button pressed somewhere on the view (plane)
* Try to manipulate the view by moving the mouse
* Nothing happens - I have to wait about 1-2 secs until I'm able to manipulate/rotate the view

(I was able to manipulate the view instantly after pressing the mouse button before/in the previous versions)

Jebtrix, 2019-08-30:
The only time I experience viewport issues in GPU accelerated software in general is from something else is robbing GPU cycles. On Mac you can use Activity Monitor to at least rule that possibility out. Or see if the slicer is pinning the GPU for some reason.

DominikPalo, 2019-09-01:
The strange thing is that I've had open both slicers next to each other at the same time in two windows:
Version 2.0.0 worked perfectly
Version 2.1.0 beta had the mentioned issue
So it doesn't seem that it was caused by system resources. Anyway,  moving the viewport works seamlessly, there were no lags etc., only the first reaction/movement is delayed.

bubnikv, 2019-09-02:
@DominikPalo Do you still have the issue with the 2.1.0-beta3?

DominikPalo, 2019-09-02:
@bubnikv yes, I have the issue also with the beta3 - see the recording (the left mouse button was pressed simultaneously with the start of the recording, but as you can see the viewport has started moving after a few seconds) [image]

bubnikv, 2019-09-03:
@DominikPalo what is your computer setup? How powerful is your computer?

I wonder what is different about your setup, as we did not receive any other report than yours on any of the platforms.

DominikPalo, 2019-09-03:
```
MacBook Pro (Retina, 13-inch, Early 2015)
CPU: 2,9 GHz Intel Core i5
Memory: 8 GB 1867 MHz DDR3
GPU: Intel Iris Graphics 6100 1536 MB
```

The output from PrusaSlicer:
```
PrusaSlicer
Version:   2.1.0-beta3+
Build:     PrusaSlicer-2.1.0-beta3+-201908301441

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 10.14.6 (Build 18G87)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version:   2.1 INTEL-12.10.12
Vendor:       Intel Inc.
Renderer:     Intel(R) Iris(TM) Graphics 6100
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod

```

DominikPalo, 2019-09-27:
I've just updated my PrusaSlicer to the final version 2.1.0 and the issue is gone! I don't encounter the issue also in RC2 and RC1, so the last affected version was beta3. Could you please describe how was it fixed? Because I don't see anything in the RC1 changelog regarding this issue, so I'm just curious whats was the problem.

bubnikv, 2019-09-27:
> Could you please describe how was it fixed?

I have no idea. Maybe something healed on your side. But it is always great to close an issue. Thanks for heads up.

closed 2019-09-27 by bubnikv

DominikPalo, 2019-09-28:
That's weird - I'm sure it was not fixed on my side because I'm still able to replicate it on the beta3 version, but not on RC1, RC2 and the final version (I downloaded and tested all of them yesterday) 🤔

bubnikv, 2019-09-30:
thanks for heads up. We are happy everything works for you now.