prusa3d/PrusaSlicer #2818: Broken feature: PVA interface

alientek, 2019-08-26:
So im very sure im not the only one that has this issue, PVA cannot bridge regardless of temperature. 

This results in a wipe tower like this:

[https://i.imgur.com/r7EG1Js.jpg](url)

The solution: Since pva cannot brigde, the way to solve this the most effecient way is to have petg or another material do the bridging. The wipe tower logic need to be changed a bit for this to work.

I have tried to explain this fix to support but they don't seen to understand to core issue here.

rtyr, 2019-08-26:
Starting with PrusaSlicer 2.1.0-alpha1 release, wipe tower was improved. Maximum volumetric speed of an active filament is respected to limit the print speed on the wipe tower. It should help with the PVA bridging problem.

alientek, 2019-08-26:
@rtyr This makes no difference, my "Maximal bridging distance" has to be set to 2mm for it to work.

Wipe tower logic needs to be changed, speed change to 15mm/s might help but is a lazy fix and increases total print time.

lukasmatena, 2021-06-03:
This issue should be fixed for 2.4.0 release ( 97c4c0200180c461a0d2504b2fe1ef6f3f6f9163 ). When soluble filament is wiped on a layer, the preceding layer does not print the sparse infill, but instead prints (almost) full layer so the soluble filament does not have to bridge.

I'll close the issue. Let us know if you consider the issue solved after first public 2.4 alpha is released (in several weeks probably). Thanks.

closed 2021-06-03 by lukasmatena