prusa3d/PrusaSlicer #2819: Supports still have an issue

murphyslawjackpads, 2019-08-26:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[](url)

bubnikv, 2019-08-26:
Please use the beta2, which was released yesterday evening. The beta has the support generator broken.

closed 2019-08-26 by bubnikv

murphyslawjackpads, 2019-08-26:
This is the beta  and still had the issues

James Williford
Murphys's Law Roadside Assistance
210-459-7743 San Antonio
512-992-9787 Austin

Closed #2819<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fprusa3d%2FPrusaSlicer%2Fissues%2F2819&data=02%7C01%7C%7Cba6ea1e3ef164e11b68508d729f6d1f5%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637024012362154416&sdata=Zo1%2BIPFsvYHtrhclZmk1GXEAxOMlGx42xmegUThtM8Q%3D&reserved=0>.