prusa3d/PrusaSlicer #2820: Direct upload to printerhost

Stomer48, 2019-08-26:
### Version
_Prusa slicer 2.0
In printersettings, IP address and API enabled. 

Use Mac OS Mojave 10.14.6
Printerhost is Octoprint 1.3.11
3D Printer used Creality Ender 3Pro; Marlin 1.1.9; No HTTPS CA file used as it is optional.

### Behavior
Direct uploading of the print-Gcode to the printer host fails.

Error message Prusa slicer 2.0: 
Error uploading to print host
HTTP 400:<html><body>400 bad request </h1>

Do I miss a setting?

Stomer48

vojtechkral, 2019-08-26:
Does the same IP work from the browser?

Stomer48, 2019-08-26:
Yes it does.

vojtechkral, 2019-08-26:
Ok, I'll have a look...

Stomer48, 2019-08-26:
Appreciate! Prusaslicer is an absolute winner!

vojtechkral, 2019-08-26:
@Stomer48 I can't reproduce the issue. Is there anything special about your setup? Are you uploading to a root directory or to subdirectory? Are there special characters in the filename?

Stomer48, 2019-08-26:
I'm puzzled! Reloaded Prusaslicer 2.0 again, and it shows the same message error. There are no special characters: CHEPCalibrationCube.gcode after slicing. If I file the slicing first, I can download it into OctoPrint with its own uploading command. It would have been nice to use the direct upload possibility via Prusaslicer.

Stomer48, 2019-08-26:
Made some screen shots:
[image]
[image]
[image]

vojtechkral, 2019-08-26:
@Stomer48 We seem to be unable to reproduce the problem. Tested on Mac OS Mojave as well, and on a Linux system, both went ok. Would you please share a 3MF file of this project, provided you don't mind sharing it?

Stomer48, 2019-08-26:
[CHEPCalibrationCube.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/3541499/CHEPCalibrationCube.gcode.zip)
Don't mind at all.

Stomer48, 2019-08-28:
Do we have to search for the problem with Octoprint?

Stomer48, 2019-08-28:
> Op 28 aug. 2019, om 12:01 heeft VojtechKra
> @Stomer48 <https://github.com/Stomer48> the file is gcode only, would you please post a 3MF file?
> Also, what does the test button in the settings tab do?

.3MF file as requested in #2820 :

vojtechkral, 2019-08-28:
@Stomer48 the file is gcode only, would you please post a 3MF file?

Also, what does the test button in the settings tab do?

Stomer48, 2019-08-28:
Seems I cannot load the .3MF file here, that is why I sent the .zip file. Please advise.
The test in the settings tab shows the same error as described before. 
Tried wifi connection as well as hard wired connection. Same results.

ChimpsterFPV, 2020-03-22:
Is there any Solution to that Problem???

bubnikv, 2020-03-22:
@ChimpsterFPV You provided a gcode, not a 3mf. 

> Also, what does the test button in the settings tab do?

You did not answer that question asked by @vojtechkral  before.

ChimpsterFPV, 2020-03-22:
> @ChimpsterFPV You provided a gcode, not a 3mf.
> 
> > Also, what does the test button in the settings tab do?
> 
> You did not answer that question asked by @vojtechkral before.

Im not the questioner, im just interested of the solution 😋

bubnikv, 2020-03-22:
> Im not the questioner, im just interested of the solution 

No problem. Then please call the Prusa3D support.

closed 2020-03-22 by bubnikv

bschollnick, 2021-06-05:
I'm running into the same issue between Windows PrusaSlicer (v2.3.1) under Windows 10, and OctoPrint v1.6.1.   Would you prefer that I open a new issue?

hockeygoalie35, 2022-04-08:
I found that when using "octopi.local/" in my address for the physical printer, I'd get error 400. After removing the slash, it disappeared and uploaded correctly.

freiform, 2023-02-08:
> I found that when using "octopi.local/" in my address for the physical printer, I'd get error 400. After removing the slash, it disappeared and uploaded correctly.

This solved the issue for me. Thanks you.