prusa3d/PrusaSlicer #2822: Wipe to object with more than 2 colors uses wipe object inefficiently/incorrectly

s4andst0rm, 2019-08-26:
### Version
PrusaSlicer 2.0.0+win64 and 2.1.0-alpha1+win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MKS3S + MMU2S 

### Behavior

* _Describe the problem_

I'm trying to reduce the size of my purge tower by using puzzle parts as wipe objects. So I setup a test project where I tried different settings.

When I'm slicing with 2 colors I managed to get the puzzle part just to the right size so that there is almost no purge tower and almost everything is purged into the puzzle part. Every second layer seems to be printed in alternating colors/filament. Purging volume is set to default .. 70 mm³ for unload and load:
[image]
So this is just fine. It would even work with less infill in the puzzle part, I just used the same amount as in the following tests for convenience.

Then I tried to do the same with 3 colors.
Obviously the purge tower gets bigger now but unfortunately the whole purge object is just purged with one color and the rest is purged into the tower. Since in the first test with 2 colors the purge tower was reduced to almost nothing I know that it is not necessary to purge that much black (into the puzzle part). And no other color is purged into the purge object:
[image]

I then tried to duplicate the purge object (I needed to set is as separate object, otherwise it would purge both puzzle parts just fully black). Now the infill of one purge object seems to be alternating white and orange layers and much less of these colors is in the purge tower. But the other purge object is completely black:
[image]

I even tried using 3 purge objects. Now the infill of two of the purge objects is purged with white and orange while the 3rd purge object is completely purged with black. The purge tower stays the same size. This tells me that even with 2 purge puzzle parts all of the needed amount is purged already, so the 3rd purge object is not needed and just additional waste:
[image]

I also tried using 4 colors and two puzzle parts just to verify that the odd amount of colors isn't the problem, but it still produced one completely black puzzle part:
[image]

* _Expected Results_

Considering the 3rd screenshot it seems to be possible for the slicer to use the perimeter of the purge object to purge one color while using the infill to purge another color. Looking at the first screenshot where every other layer is a differt color while purging almost nothing into the purge tower it seems to me that the puzzle part with its infill is large enough to be able to purge almost all of the amount needed for one color.
When using 3 colors one color is used in the next layer again immediately after the layer change so only very little of this color needs to be purged in the current layer (for example in the 3rd screenshot a lot of white is purged but only very little orange, while in the next layer a lot of orange is purged an almost no white).
So a part of black is purged to the perimeter in every layer while orange and white always is purged into the infill. But the rest of black could be purged into the purge tower. Instead I currenlty need to a make a sacrificial additional purge part where it can purge the rest of the black.
Even when I increased the size of the single purge object to 300% so the perimeter definitely is long enough to be able to purge all of the black filament nevertheless the whole purge object is black and no while/orange is purged into the infill of it. 

So my expected result would be that when using a single puzzle part as purge object, black is purged maybe to the perimeter and additional black into the purge tower if the purge object's perimeter isn't long enough to be able to purge all of the black. And white and orange would be purged into the infill of the purge object alternatingly. And no second purge puzzle (which is all black) is needed.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
3MF with clownfish and puzzle parts attached (2nd puzzle part disabled).
[clownfish_purgetest.zip](https://github.com/prusa3d/PrusaSlicer/files/3540151/clownfish_purgetest.zip)

lukasmatena, 2019-08-26:
Please, next time enclose a 3MF and specifically state what is wrong when slicing that very 3MF. It is quite difficult to make sense of it. And maybe I did not made it right.

> So my expected result would be that when using a single puzzle part as purge object, black is purged maybe to the perimeter and additional black into the purge tower if the purge object's perimeter [...]

Why do you care about whether it is purged in the infill or the wipe tower? It has to be purged somewhere and you designed your object to be just big enough to purge ONE color. The rest will go to the wipe tower. Black, orange or white, the wipe tower will grow. What is the difference?

> And no second purge puzzle (which is all black) is needed.

It is not 'needed'. It is only needed if you want to purge into something else than the wipe tower. If there are two toolchanges on a layer and you want to purge 140 cubic mm for each, you need to have at least 280 cubic mm on sacrificial objects extrusions. If you don't, something has to go to the wipe tower, regardless of the purge order.

Also, the wipe into infill can really be suboptimal on objects with very large single island with dense infill. The algorithm cannot split an infill island into two, so if it needs to purge 140 cubic mm and there is an infill island that needs 200 cubic mm, it will purge more. That may (but probably isn't) part of the problem.

If I misunderstood you, I'm sorry. In that case please follow my advice from the first line. It is of course possible that there is some bug.

s4andst0rm, 2019-08-26:
I didn't really have an imagination of how much perimeters the puzzle part would need in order to be to be able to purge 140 mm³ into its perimeters. It seems it would have to be much larger than I expected and that's why it only worked, when I set a whole separate object additionally, because now all of the black filament could be purged into the second purge object and the rest of the black went into the perimeter of the other wipe object while the remaining 2 colors were wiped into the infill of the purge object. 

I sliced only the puzzle part with 0.2mm height in a separate project with no top/bottom/infill and increased the perimeters until the whole file had more than 140 mm³. That seems to need 10+ perimeters. Is this the best way to determine the needed size? Or can I somehow see in PrusaSlicer how many cubic mm are purged into the perimeters and/or infill of the purge objects?
I did the same with no perimeters and with 50% infill and got around 140 cubic mm .. so this seems to be the minimum needed amount of infill.

I then worked my way back in the project with the clownfish:

When I use 9 perimeters on the purge puzzle object and 50% infill, most of the black filament is purged into the perimeter and orange and white is purged into the infill:
[image]

If I decrease the perimeters to 8, the whole purge object is purged with only black:
[image]

The main difference is that in the first case the total amount of used filament is ~41m and only ~9m go into the purge block. In the second case ~47m are used in total and 16m go into the purge block.

> Why do you care about whether it is purged in the infill or the wipe tower? It has to be purged somewhere and you designed your object to be just big enough to purge ONE color. The rest will go to the wipe tower. Black, orange or white, the wipe tower will grow. What is the difference?

I want as much of the purged filament to be in the purge object, because I can use these puzzle parts to decoratively put them together while the rest which is in the purge block more or less just goes to the trash bin. So the goal is to make the wipe tower as small as possible and preferably purge into the purge object.
So by choosing the exact right amount of perimeters in my example I can reduce the amount of total used filament by ~6 meters.

---

So I guess my initial problem changed somewhat now that I identified my initial lapse in thought with your help:

It would be nice if the purging process could split colors/filaments on the whole purge object within a feature (perimeter, infill) so in the above example with 8 perimeters and maybe 60% infill (so there definitely is enough infill for 140+ cubic mm) the black could start with the perimeter and continue a little in the infill and purge the rest of the infill with the orange/white filament and only purge the rest into the purge tower if the perimeters and infill are still not large enough. This way one would not have to exactly have the right amount of perimeters in order to save some filament.
Would this be possible?

> Also, the wipe into infill can really be suboptimal on objects with very large single island with dense infill. The algorithm cannot split an infill island into two, so if it needs to purge 140 [...]

I don't exactly understand what you mean. Can you please elaborate?
Is the whole perimeter considered as an "island" and is this exactly the root of my problem?

---

3MF for screenshots above included.
[clownfish_purgetest2.zip](https://github.com/prusa3d/PrusaSlicer/files/3541191/clownfish_purgetest2.zip)

lukasmatena, 2019-08-26:
> Is the whole perimeter considered as an "island" and is this exactly the root of my problem?

If you have a simple object like this, there is essentially only one island and the wipe into object feature has only two places where to wipe - its perimeters and its infill. And it cannot split either of these, even if it is larger than needed. This can be observed when you scale your puzzle piece so it's really large and set number of perimeters to 30 or so.

> It would be nice if the purging process could split colors/filaments on the whole purge object within a feature (perimeter, infill)

It surely would, but we are likely not going to make such a change anytime soon. That would probably be a lot of work with high risk of breaking something. I just thought of a workaround though. Put a modifier mesh inside your object and set at least one of its properties so it differs from the enclosing volume. This effectively breaks the infill into more logical entities. I'm attaching a 3MF and a screenshot so you know what I mean. This is quite nasty, but it should work. (Of course you can just add one big modifier mesh to enclose a half of the object if you know you want to split in half.)

[image]

[clownfish_purgetest_workaround.zip](https://github.com/prusa3d/PrusaSlicer/files/3541479/clownfish_purgetest_workaround.zip)

s4andst0rm, 2019-08-26:
> If you have a simple object like this, there is essentially only one island and the wipe into object feature has only two places where to wipe - its perimeters and its infill. And it cannot split either of these, even if it is larger than needed.

Ok, thank you, that's what I thought.

> It surely would, but we are likely not going to make such a change anytime soon. That would probably be a lot of work with high risk of breaking something.

Yeah, I can see that. 

But now that I know more of the internals how wipe to object works, I maybe can come up with another approach which fits my needs.
Your workaround looks like a good way to start. Thank you.

closed 2019-08-28 by s4andst0rm