prusa3d/PrusaSlicer #2823: Distance from Raft as Air Gap like in cura

MThoilir, 2019-08-26:
### Version
PrusaSlicer 2.0.0

### 3D printer brand / version + firmware version (if known)
CR-10
soon I3 MK3S

### Behavior
When i want to use a raft to give my option very good adhesion. But i want a good bottom surface of my print. I dont get these with PrusaSlicer, the printed lines on the first layer dont even stick to each other and dont stick to the next layer which is fine. But with cura it works all fine and i have a compact, a bit structured, bottom layer. 
The difference ist that cura uses called "Air Gap" instead of distance.
You set the Air Gap to 0.3 mm. So it prints the first layer of the object in a height of +0.3 mm, so u can detach this layer from the raft after the print. For the next layer position it substract this 0.3 mm and prints it.
That means for the first layer of the object its 0.3 mm heigher. But all following layers have the same height like printed directly on the raft without gap.
In addition, u synchronize the layers of support and object. Because the support is printed directly on the raft.
Please add the option Air Gap directly to the raft layer count in PrusaSlicer.

I know, that i am not a printing profi and my xyz cube test maybe has some other issues. And the cubes where not intentionally made to show this here. But i hope, it works for you.

I think this post is new, because i couldnt find it in the search bar. But i am new to github too. So please forgive me if this is already mentioned or maybe only i have this problem with my CR-10 and its not a problem for a Prusa printer. But i want to buy a Prusa with MMU2 and only use one slicer then.

[image]
[image]
[image]
[image]

MThoilir, 2019-09-04:
can someone show me a plane surface coming from a raft on a Prusa Printer and sliced by PrusaSlicer? Thanks

bubnikv, 2021-02-24:
Implemented with #6003 and aee136cedd890ecb5072d9c9215a824fa3f365b1

Object over raft is no more printed with bridging flow, but with a normal flow.
Gap between the raft and 1st object layer is configurable independently from the support gap.

closed 2021-02-24 by bubnikv