prusa3d/PrusaSlicer #2824: Arrange on Custom STL Bed

foreachthing, 2019-08-26:
### Version
2.1.0-beta2+win10x64

### 3D printer brand / version + firmware version (if known)
Ulitmaker 2

### Behavior
* Add custom Bed (stl attached)
* add parts
* click arrange
* -> parts not where expected

Result:
[image]

Expected:
[image]

#### Project File (.3MF) where problem occurs
[CustomBed.zip](https://github.com/prusa3d/PrusaSlicer/files/3541506/CustomBed.zip)

tamasmeszaros, 2019-08-27:
Hello! I could not reproduce this exact behavior but the arrange feature is not yet optimized for custom shaped beds and can produce poor results. 

If this happens you have no option but to use the rectangular bed, set to the real usable print area dimensions. In the meantime I will improve the irregular bed arrangement but I'm sure it will be much slower compared to the rectangular bed version.

foreachthing, 2019-08-27:
No worries! I just felt like reporting this _issue_.
I use the rectangular until it is fixed - or not fixed :-)
Maybe I just draw the boundaries onto the svg....

JohnOCFII, 2019-09-01:
Can you use a custom image, but still use the standard rectangular shape for arranging?  The areas of my custom bed image aren't areas where I can place items anyway, similar to the default Prusa bed image that had area for the heater connection, lift tab area, etc.
[image]

barry99705, 2019-09-01:
[image]
Here's what it's doing with my custom bed stl.  I need to avoid the two mounting tabs on the side to clear my hot end fan.  I can manually arrange my parts, but just hitting the button is so much easier!  Seconded for John's idea, that would be handy.

bubnikv, 2019-09-02:
I checked with @enricoturri1966 and @tamasmeszaros . @tamasmeszaros developed the arrangement algorithm, @enricoturri1966 implemented the custom print bed.

The print bed definition consists of the following:

1) The working space shape (Rectangular, Circular, Custom). Custom can be defined by loading an STL and extracting its contour.

2) Texture. The texture is mapped over the bounding box of the shape defined in 1). If you have a working space (defined in 1) with cut outs, the rectangular texture is mapped over the cutouts.

3) The model. If loaded, the texture is rendered centered at the [0, 0, 0] point of the model, scaled by the bounding box of the working space shape defined in 1).

Now the arrangement is performed over the working space shape defined in 1). According to @tamasmeszaros, it is much more computationally demanding to arrange over complex working spaces as this rectangle with cut outs, and we did not put too much effort into it. For now, we advice you to define a simple rectangular working space in step 1).

foreachthing, 2019-09-02:
Suggestion for others (not devs):
Use a rectangular/round shaped bed and draw your **own texture** (I did mine in Inkscape to get exact measurements) with all the safe-areas. Then, arrange will work and if something would fall out of your safe area, you'd see it and could move those parts to a safe location.

JohnOCFII, 2019-09-02:
Thank you for the clarification.  It seems following @foreachthing's idea is the way for me.  I was a bit unclear given the three options displayed and how to use them.  Some tool tips might help others avoid this performance issues if all they are looking for is a texture.

[image]

bubnikv, 2019-09-02:
>  Some tool tips might help others avoid this performance issues if all
they are looking for is a texture.

I think the biggest problem now is the lack of manual. The content team is
working on it.

closed 2020-03-16 by tamasmeszaros

foreachthing, 2020-03-16:
@tamasmeszaros  uhm ... why was this issue closed?
It's still not working prperly (version 2.2.0-rc4; [built](https://github.com/prusa3d/PrusaSlicer/commit/c1e92a575951069c354405cc1e9984d38e4339b8)).

See 3mf attached. Somewhere along the 26th, 27th instance, something goes wrong.

[image]

[Rad.zip](https://github.com/prusa3d/PrusaSlicer/files/4337480/Rad.zip)

tamasmeszaros, 2020-03-24:
@foreachthing The issue was closed because I think it had a positive resolution in a form of a workaround. The arrangement for irregular shapes is a feature that has to be addressed with a complete dev cycle (not only a bugfix) and its definitely on our todo list.