prusa3d/PrusaSlicer #2825: Add the length of each filament used

Regaor, 2019-08-26:
Hello and sorry for my bad English, I am French :D 

Would it be possible to add the length of each filament used in addition to the total length? It would be very convenient to finish almost empty spool!

Thanks a lot
Regis

[image]

sgano, 2019-12-31:
I agree - this would be really useful especially considering all of the new Print Color capabilities in V2.2.0-Alpha.

wpegden, 2020-02-18:
I agree with this also.  Currently I load the gcode into a separate gcode viewer just to get this information.  Clearly the individual lengths are already available as they must be used to figure the cost estimate.

BhaaLseN, 2021-04-04:
+1

I just did a print with color change, and was a bit disappointed that it only showed me the total amount of filament being used; but not on a per-filament basis. This makes tracking how much is left on the spool a bit difficult.
The Color Print view shows the time per color (plus the time when the change occurs), but nothing regarding filament. It would be nice if that info could be put into columns up there at the top:
[image]

rtyr, 2021-06-14:
Implemented on master. It will be included in PS2.4.0-alpha1. Closing.

[image]

closed 2021-06-14 by rtyr

BhaaLseN, 2021-06-14:
> 
> 
> Implemented on master. It will be included in PS2.4.0-alpha1. Closing.
> 
> [image]

Does this also apply when using filament change on a single-extruder machine? Doesn't look like it from a quick glance at the code (unless filament change is handled internally as a virtual extruder).

rtyr, 2021-06-14:
Yes.

[image]