prusa3d/PrusaSlicer #2829: Setting negative first layer extrusion width results in wrong first layer extrusion width

Goury, 2019-08-27:
### Version
PrusaSlicer-2.0.0+linux64-201905201652

### Operating system type + version
Ubuntu 18.04 with 4.15.0-58-generic kernel

### 3D printer brand / version + firmware version (if known)
Custom Kossel with Duet3D

### Behavior
Run Slic3r, go to print settings → advanced
Set extrusion width → first layer to some percentage under 100%
Now it slices the first layer in super thin lines and it won't work because my machine will extrude so little it won't adhere to my bed.

### Expected behavior
Extrusion width of 90% for the machine with 0.4mm nozzle to be equal to 0.36mm setting.
If it works when I input 0.36 why does it not work when Input 90%?

#### Project File (.3MF) where problem occurs
[The toolholder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3543613/The.toolholder.3mf.zip)

Goury, 2019-08-27:
Hmm.. Okay, it really does say that.
But five lines long tooltip popup with no formatting isn't very readable.

So I'd like to change that to feature request.

Taerog, 2019-09-09:
Agree, this is quite confusing and everyone so far (I have seen using the beta) assumes its a % of nozzle Both intuitive and like all other slicers.   If there is a reason for this real clarification and final computed number needs to be displayed, otherwise possibly revert to % of nozzle width

bubnikv, 2020-02-04:
> otherwise possibly revert to % of nozzle width

Slic3r always worked that way, it always calculated extrusion width over layer height. If we change that, we would first need to change a lot of code, then we would have to explain to the current users why their profiles are broken.

Goury, 2020-02-04:
@bubnikv sounds like a good thing to do for the 2.0 release.
Point for ineffective project management.

Just because something was counter-intuitive from the beginning and is deeply burrowed into the code isn't the reason for not fixing the issue.
So isn't the backwards compatibility, but here's an idea for you: make it so %{option_name} makes it relative to any named option and keep layer height as default when no option name is given.
This way old configurations will behave as before but we'll have a clear option to configure it in a sensible way.

towe96, 2020-03-29:
I'd still love to see this **bug** get fixed. Extrusion width over layer height seems completely useless.

In /PrusaSlicer/src/libslic3r/Flow.cpp, I'd expect the code at [line 90](https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/Flow.cpp#L90) would need to be changed to something like this:

`	if (opt->percent) {   
    auto opt_nozzle_diameters = config.option<ConfigOptionFloats>("nozzle_diameter");
    if (opt_nozzle_diameters == nullptr) {
    		throw_on_missing_variable(opt_key, "nozzle_diameter");
    }
    double nozzle_diameter = opt_nozzle_diameters->getFloat();
    return opt->get_abs_value(nozzle_diameter);
	}`

I couldn't get it to compile for other reasons though - never fiddled with VS before.

towe96, 2020-06-21:
It has been fixed in the "SuperSlicer" version:

https://github.com/supermerill/SuperSlicer/blob/master/src/libslic3r/Flow.cpp

bringbackgroupb, 2020-09-12:
There are a few issues going on here. The first is that the tool tip is not properly worded in English. Saying something is "calculated over X" doesn't really mean anything to most native English speakers. A better way to say this would be "calculated from X". I was confusion about this until I read through all these threads about this issue.

Secondly, a lot of us design our parts around extrusion width. I always print with either a 0.4mm nozzle @ 0.5mm or a 0.8mm nozzle @ 1mm width. This means that the walls in my parts are tuned for these parameters. See how they're also multiples of each other? This means parts I design can be printed on either printer (most of the time).

Lastly, extrusion width should be tied to printer, not to filament or print settings since it is a function of the physical hardware installed on the printer, not a software configurable parameter. I should not need to build print profile settings for each nozzle and material. If it's going to be like that (like in S3D), the just combine all settings into one long list instead of breaking them up as Print/Failment/Printer.

ardenpm, 2020-11-27:
This definitely becomes a big issue on Tool Changing printers with different nozzle sizes on each tool. While being able to use percentages relative to nozzle instead of layer would help that, there might still be cases where you want precise absolute control over extrusion width, per extruder/tool.

neophyl, 2020-11-27:
On multi tool systems its not just a few parameters from the print profile that you might want to use either.  Its all of them.  
At the moment you can assign different filament profiles to different tools but the Print profile applies to all tools.  This means you have to use 0 for extrusion widths to get any practical use for the different nozzle sizes defined for different extruders.

It would be beneficial if you could assign separate print profiles with all the settings included to the different tools like you can with filaments.

towe96, 2020-11-27:
There's just no sane reason to ever base extrusion width on layer height, especially when considering the "variable layer height" feature.

ardenpm, 2020-11-27:
> There's just no sane reason to ever base extrusion width on layer height, especially when considering the "variable layer height" feature.

I personally agree, however the issue now will be that changing it would break compatibility and existing profiles and I suspect that will be the issue the devs would raise with proposed changes that reinterpret the existing meaning. So my guess would be that percentage being layer height would have to remain the default and a way to allow it to be over nozzle size instead could be enabled by an option (e.g., a checkbox for that purpose).

bubnikv, 2020-12-15:
> I personally agree, however the issue now will be that changing it would break compatibility and existing profiles and I suspect that will be the issue the devs would raise with proposed changes that reinterpret the existing meaning. 

Exactly. We don't have the luxury of changing a meaning of a parameter on a whim.

Goury, 2020-12-15:
Oh, but you do.

Firstly, absolutely no one is using this setting is for this exact reason: it is totally unusable and it doesn't make sense to use it for anything.

Laslty, you have write access to this repository.

Kachidoki2807, 2020-12-18:
@Goury, there are so many things in this world that should have already been changed just because it makes sense. Unfortunately it is not always possible. Sometimes it is possible, but at the cost of an effort that not everyone wants to provide. I let you find a lot of obvious examples.

@bringbackgroupb 
> Lastly, extrusion width should be tied to printer, not to filament or print settings since it is a function of the physical hardware installed on the printer, not a software configurable parameter.

It is not so obvious. The print settings **needs** to be adapted to the nozzle size or hardware. For instance, you need to adapt the speeds to the nozzle size or the capacity of your machine, you can't only rely on the MVS from the filament setting to get good results. The supports settings also needs to be adapted, even more finely on big nozzles. The filament profiles also needs to be adapted to your nozzle, depending on the material of the nozzle, the temperatures are not the same.

> I should not need to build print profile settings for each nozzle and material. If it's going to be like that (like in S3D or Cura), the just combine all settings into one long list instead of breaking them up as Print/Failment/Printer.

It is not easy nor obvious at all to find an universal way to arrange smartly all the settings. Exposing them completely into a single list is ok once for a single specific setup, but when you want to have more flexibility, it become a burden to have to check, let say, all the accelerations to be sure there are still adapted to the printer you want to use. I often use the same print settings on three different printers (all are i3 but don't have the same bed size nor acceleration for instance). When I fine tune the print settings (like infill, speeds or even extrusion widths) for a parallel production, I am happy to have to change them only once for three printers.

As a developer, I fully understand the reluctance of changing such a deeply implemented parameter, even more if it is used into the users custom profiles. In the meanwhile of finding a proper solution (and time!) I suggest the side text "mm or %" to be more explicit to not hide this ambiguous meaning under a tooltip. It is a workaround, but better than nothing.

bubnikv, 2021-01-20:
> Oh, but you do.

What?

> Firstly, absolutely no one is using this setting is for this exact reason: it is totally unusable and it doesn't make sense to use it for anything.

> Some users are using the "width in percent of layer height" logic, see #4631. 

> Laslty, you have write access to this repository.

So?

bubnikv, 2021-01-20:
Duplicate of #1923

Photogad, 2021-02-12:
> Exactly. We don't have the luxury of changing a meaning of a parameter on a whim.

@bubnikv 

It would only break profiles for people using percentages in the width settings, and I doubt many user are, because setting widths as a percentage of first layer **height** makes absolutely no sense.

Widths should only matter to the diameter of nozzle you are using, never layer height...

I still run a 0.20mm first layer on my 0.80mm nozzle but that doesn't mean I want the default extrusion width to be 0.10mm (50% of 0.20mm) on a 0.8 nozzle, lol.

The code is already in SuperSlicer, just pull from that fork or maybe @supermerill  can do a pull request

towe96, 2021-03-22:
Any updates on this _bug_?

Yes, this would break compatibility. All 0 users of the old logic of this feature will be affected.

VanLaser, 2021-07-30:
I don't understand, how is this a bug? If you want to keep the _shape_ of the paths (same 'thickness', i.e. oval shape, good for bonding) _when changing layer height_, this feature can be very useful. This breaks people expectations, alright, and also if they don't configure a maximum volumetric flow, their printer may go too fast and may not be able to extrude when the height (and thus, the extrusion width - and volume) is high. But, its logic in itself makes perfect sense. Maybe the tooltip should have more details, or a warning about capping max volumetric flow, or just say 'advanced, don't use unless you understand it' etc.

Goury, 2021-07-30:
@VanLaser how is it not?    
This feature is utterly useless for keeping parts shape when changing layer height.    
It literally does absolutely nothing in this case.    
It changes first layer extrusion width to a percentage of its height.    
Which is not useful for anything and is just purely destructive.

Efficient extrusion width depends on nozzle width.
Efficient extrusion height depends on combination of the bed and material.
Extrusion volume is calculated from above two values.

But in slic3r we have no way to use nozzle width as a factor, thus making automated calculation of the first value impossible.

VanLaser, 2021-07-30:
This is not about 'first layer', it's about all layers. 

Also, I'm not trying to make this a right vs wrong, just trying to explain something.

Depending on prints, I often prefer to change my layer height. Sometimes smaller, sometimes larger, sometimes variable. But I don't change my nozzle width too often.

For that chosen layer height (not only the first layer), I want the shape of the melted filament section/path to look a certain way, an oval shape, which helps with bonding, which means, the _width of that shape will depend on layer height_. For the external walls, maybe I want a shape close to a circle, to look better. It's counterintuitive that height and width would go together, but (if you read the first link above), it's the shape that you want to model, not some intuitive compromise. 

https://manual.slic3r.org/advanced/images/flow-math/area1.png

The extrusion volume will depend on that surface (width * height) and the speed of the extrusion on that layer. 
All this works if you limit the total extrusion volumetric speed in prusaslicer so it will slow down automatically when needed and not go over that limit. What this means is, practically:

I can push more filament, but slowly, and make higher (and larger) layers, or: I can push less filament, with smaller layers, but faster. In both cases, I decide the shape of the layer (how 'oval' it is). I can do that in prusaslicer, using these three settings (layer height, extrusion width as percentage of layer height, max volumetric speed limit). You don't have to limit all the other speeds, unless you really want to, and prusaslicer will compute them for you, trying to keep a close to max volumetric flow, which is ... optimal, actually.

As long as the extrusion width depends on the layer height, using percentage values feature, I can do that, and I am doing that, so it is not useless. But, in the past, I did not see things at all like that, until I read about it. Like the links from my previous post.

Goury, 2021-07-30:
@VanLaser it is about first layer and also about you not being able to read both setting name, tooltip and this very thread.    
For your case you don't need any additional settings, slic3r does adjust extrusion according to your layer height with combination to your nozzle width.    
The setting we are discussing in this topic have absolutely nothing to do with any layer past very first.    

As for your particular case, it's still a bad idea to take nozzle width out of equation.    
Even if you want your extrusion width to depend on layer height for some reason, it must be a factor of nozzle width, not just layer height regardless of hardware.    
Because a machine can have more than one nozzle and those may be different.    

The problem here is that Joseph doesn't want to support anything other than what he sells and is actively blocking support for features required for slic3r to be useful to other machines.    
This is why other forks exist.

VanLaser, 2021-07-30:
Oh wow. I think this 'dialogue' is quite asymmetric, with one person providing some info and the other one patronizing.

I suggest maybe to visit the PrusaSlicer extruder width section and read all the other tooltips, too. Your problem may be related to the first layer, but the other extruder width settings are consistent, they do apply to other layers. Also read the title and the contents of your own original post, you don't ask for changes in first layer settings, you simply ask to change 'extrusion width'.

SuperSlicer changing this setting actually makes me think twice before using it.

Also, I think I won't bother to continue participating in this thread.

towe96, 2021-07-31:
"Extrusions must be oval to stick" is just not a thing, which is why "some info" doesn't really matter here.
You'll need to constrain volumetric extrusion speed regardless of line width settings anyway.

It's counter intuitive and serves no useful purpose as a settings, which is why 0 people would be negatively affected by finally fixing this **bug**.

And I'll keep calling it a bug because of Hanlon's razor.

VanLaser, 2021-08-06:
You can do whatever, I will stick to 'some info' as opposed to "it's not a thing" which is zero info. To remind you too, I chimed in here just to say the setting makes sense to me, and it's useful, since I prefer to have that oval shape which made sense according to that 'info', and it works for me. In other words, at least one person found that it can use the setting as is, and likes it. A part of the oval shape, zero population, apparently.

Call it a bug until cows come home, regardless of what I wrote, I remember github being a place where you really have to present good justification for your issues, not just say 'it's useless', 'it's not intuitive', 'it doesn't matter' and so on. 

Also the original 'first layer' issue would have more sense to me to be addressed as an extension of the 'big foot compensation' setting, which now accepts a mm value and it modifies first layer width with it. That could be naturally be extended to also accept a percentage of the same first layer width.

Goury, 2021-08-06:
A good justification: existence of this feature makes people to misuse it for no good reason.

VanLaser, 2021-08-06:
Bravo! Again, a very facebook-3d-printer-group-haha-meme 'smart' answer, but not actually helping your cause. Try reading some github issue descriptions from other, non 3d-printed software repositories. Or just blame the developer, that also help you (not doing any work).

towe96, 2021-08-06:
Well it's just the oval shape doesn't make sense, thus your unique use of the feature isn't really a reason to keep this bug around.

ardenpm, 2021-10-30:
Leaving aside the debate for a moment, I think at the very least there should be the ability to disable this warning.

[image]

Since it is much more common to use mm you are presented with this basically every time you change an extrusion width, and there are now eight of them that can be changed.

supermerill, 2021-10-30:
> I think at the very least there should be the ability to disable this warning.

I just added a "max_literal" field to the config def, so each setting can set its threshold for which this pop-up (very high value disable it) supermerill/Superslicer@ff195fc9

It's an idea.

bdistin, 2022-07-24:
Just came from superslicer to try out arachne... Man this was a bad user experience to track down. It looks like you don't want to break the current behavior, but have you considered adding a checkbox to allow the nozzle diameter based behavior? Seems like a happy medium, don't break anyone by default, but allow smoother transitions from superslicer and allow this more adaptive behavior for different nozzles without creating profiles for every different nozzle diameter.

nick-parker, 2024-04-01:
If we insist on maintaining compatibility, why not add an alternative symbol for percentage of layer height?

eg:
layer_height: 0.2
nozzle_diameter: 0.4

extrusion_width: 250% -> 0.5 mm
extrusion_width: 125N -> 0.5 mm

I don't love the notation because without the percentage sign it's a bit unclear whether I ought to write 1.25N or 125N. Maybe Just make the symbol %N or something like that.

But in general, this is a text field so we have a lot of freedom to sort this out. I don't like the idea of a checkbox because that would either mean lots of checkboxes or locking "what does a percentage mean" across all the width fields, which seems confusing.

If there's support for this approach I'm happy to implement and PR. This would be helpful for our Hylo profiles at AON3D.

lukasmatena, 2026-09-01:
Changed in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena

Goury, 2026-09-01:
Oh wow.
Finally.
Took you just a bit over seven years.