prusa3d/PrusaSlicer #2832: Baudrate issue on Mac connection prusa printer to flash firmware

olhodneland, 2019-08-27:
### Version
2.0.0+

### Operating system type + version
MacOS Mojave version 10.14.6

### 3D printer brand / version + firmware version (if known)
Mk3s and Mk3 

### Behavior
When connecting the printer with USB to Mac to be able to fash new firmware, the port shows up in the "firmware flash menu", but when it starts to flash it fails almost every time. It says its a timeout issue or something.
If Mac is connected with usb, without trying to flash, the printer starts over and over again. An typical baudrate issue.
I use repetier server on a raspberry pi 3B+ normally, to get to control the printer over wifi. I tried to connect to the Raspberry pi instead to flash the firmware from there. I noticed the same problem. The printer started over and over again. In repetier server I had the baudrate set to auto detect. When I changed it to fixed 115200 it worked right away. 
This was kind of a confirmation of my baudrate suspicion.

So what i suggest you to do, is to look into the baudrate settings in prusa slicer on mac. This shoud be just fixed to 115200 or at least be possible to choose from a menu.

Please give a feedback what you think. 
This issue has happend several times when trying to flash new firmware, and results in the need of full factory reset.

vojtechkral, 2019-08-27:
> If Mac is connected with usb, without trying to flash, the printer starts over and over again. An typical baudrate issue.

Baudrate issue seems unlikely to me. Sounds more like that either the cable is flaky or there's something in your system repeatedly opening serial ports. There is sometimes such software on some Linux distros, not sure if the same can be the case on Mac. Did you try with a different USB cable? Does your USB cable have a [ferrite bead](https://en.wikipedia.org/wiki/Ferrite_bead)?

> So what i suggest you to do, is to look into the baudrate settings in prusa slicer on mac. This shoud be just fixed to 115200 or at least be possible to choose from a menu.

Currently the MK3 flashing baudrate _is_ fixed to 115200 on all platforms (see [the code](https://github.com/prusa3d/PrusaSlicer/blob/a1f34adcf6b4e48f00056b00fea534b5af5649a5/src/slic3r/GUI/FirmwareDialog.cpp#L484) ... there is a todo note considering options for other baud rates but so far this didn't seem necessary, so it just uses 115200).

olhodneland, 2019-08-27:
But the exact same thing happend on the raspberry pi until i fixed the baudrate. When connecting the usb to the raspberry pi the printer just started over and over again. The same i have experienced when choosing wrong baudrate, when i have built multiple selfmade printers. 

I used the USB cable following the mk3s in the package every time.

vojtechkral, 2019-08-27:
@olhodneland what software did you use to flash from the raspberry? Which badurate was the wrong one?

olhodneland, 2019-08-27:
I used repetier server on a raspberry pi 3B+. And it was set to auto detect on baudrate. So not sure what baudrate causing the problem.

olhodneland, 2019-08-27:
@vojtechkral I tried prusa slicer first but that did not work, as said. Repetier server did work

vojtechkral, 2019-08-27:
@olhodneland Could you please post your firmware flasher log with the failure? The baud rate can be seen in the log, too...

olhodneland, 2019-08-27:
Then I have to try to flash again from Prusa Slicer. Dont have the time for it at the moment because I need both printers for work. But i can try again when I get time.

bubnikv, 2019-08-28:
> There is an issue with Autobaud detection with some Prusa models:
>
> Cannot negotiate baud rate connecting to Prusa i3 MK3
> foosel/OctoPrint#2517 <https://github.com/foosel/OctoPrint/issues/2517>

There is no auto detection at our firmware side, the baud rate is fixed to
115200.

rtyr, 2021-02-03:
No response. Closing.

closed 2021-02-03 by rtyr