prusa3d/PrusaSlicer #2835: Error message: The Wipe tower is only supported if all objects have the same layer height profile

s4andst0rm, 2019-08-28:
### Version
PrusaSlicer 2.1.0-beta2 and PrusaSlicer 2.1.0-alpha1

### Behavior
* _Describe the problem_

When having two parts with a different extruder assigned (doesn't matter if separate objects or single object having multiple parts) and using the "Edit Layers" feature afterwards the program cannot slice anymore and throws the error "The Wipe tower is only supported if all objects have the same layer height profile" even though a 0.2 layer height profile is used and in the layer editing properties also 0.2 layer height is set:
[image]

* _Steps needed to reproduce the problem:_

- Add any 2 objects to the workspace with a MMU2S printer profile
- Assign different extruders to the 2 parts
- Right click one object and select "Edit Layers"
- Enter Min Z and Max Z (or skip, doesn't matter)
- Slice
- --> Error message is shown

#### Project File (.3MF) where problem occurs
3MF attached
[layer_editing_error.zip](https://github.com/prusa3d/PrusaSlicer/files/3550788/layer_editing_error.zip)

bubnikv, 2019-08-28:
That is an expected behavior. We do not allow that, as the current tool
exchange over wipe tower only works if all layers of all objects are
synchronized. If they were not synchronized, the amount of waste material
and the size of the wipe tower would be excessive.

s4andst0rm, 2019-08-29:
All layers are the same layer height. The print settings are using 0.2 layer height and in the layer edit options 0.2 is selected. They are synchronized in my optionion. Or do you mean something different by 'synchronized'?

I want to use the layer edit feature to change the infill density in a specific layer range. I thought that's exactly what the edit layer functionality is for.

As a workaround I can add a generic box as a modifier, resize/transpose it to the desired size/location and change the infill with that.
How does this differ regarding the generation of the wipe tower compared to the layer editing functionality?

s4andst0rm, 2019-09-09:
I guess I misunderstood the "edit layer" feature (or "Height range modifier" as it seems to be called now in the 2.1.0-rc): I thought it could also be used to change the infill etc. for a specific layer region, because you can click the gear-icon to modify some options:
[image]

In a printer profile without MMU (or wipe tower to be specific) it DOES work like this. I have to specify a layer height (or leave it as default so it matches the layer height in the rest of the print) and can then change the infill etc. in the specified region.

So I guess I'd like to change this issue to a feature request:
I'd be nice to be able to use the edit layers / height range modifier in an MMU print just to change the infill etc. for a specific layer range. Of course this requires that the layer height in the edit layers settings is set equal to the layer height in the printer profile (which is default anyways). This makes changing the infill of a specific layer range way easier compared to a generic box added as a modifier because if you have scaled the object you have to do a lot of calculations to get it to the right height whereas in the edit layers settings you can just enter the absolute values (in mm).

lukasmatena, 2019-09-09:
@s4andst0rm I don't think you misunderstood anything. What you reported is a bug - the validation should be able to see that the layer height was not actually modified and allow slicing in that case.

We'll fix this, but not for PrusaSlicer2.1.0 release. The fix will be included in 2.2.0 or a patch release of 2.1.0 if there is one.

s4andst0rm, 2019-09-09:
@lukasmatena Ok, thank you very much for clarification.

lukasmatena, 2019-09-17:
This should be fixed by b43003d. The fix will be part of the next release PrusaSlicer2.2.0.
Closing.

closed 2019-09-17 by lukasmatena